logisim-evolution/logisim-evolution #1050: Feature request: Copy-paste over different logisim instances/projects

BFH-ktt1, 2021-09-06:
Currently logisim allows only copy/paste/duplicate in the same project space (internal clip-board). Kevin Walsh implemented in his holy-cross version a possibility to copy-paste over different projects using the systems clipboard. This feature would be very nice to have. Back-porting of this feature might be more effort as setting it up from scratch.

MarcinOrlowski, 2021-09-06:
If every element can be serialized for saving, we can pass that representation via clipboard.

BFH-ktt1, 2021-09-06:
Yes, all can be serialized (as been done in xml in the .circ file) Kevin Walsh used this same format (xml) to do the copy-paste

BFH-ktt1, 2021-09-06:
What might be more an issue would be a copy in logisim, and for example a paste in a drawing program, than there needs to be an extra writer attached (like the one's used to do picture export in the different formats)

MarcinOrlowski, 2021-09-06:
We should export SVG. That'd be be the best option.

R3dst0ne, 2021-09-06:
Is this a duplicate of #56?

BFH-ktt1, 2021-09-07:
Yes this is correct. Closing #56 as we have a link here

maehne, 2026-09-07:
@hewzhew: Thanks for your analysis! I agree that it makes sense to address this enhancement in multiple steps as outlined by you above. 

Where it makes sense, using/adopting @kevinawalsh's approach (with proper reference) may make sense to limit code divergence to facilitate later back-porting of other enhancements by him. If none of his relevant commits can be adopted/merged, noting these commits as discarded in the PR may help us, too. However, you don't need to spend too much effort on this aspect.