logisim-evolution/logisim-evolution #1054: TTL library item 74163 not clearing output on high to low transition of CLR pin.

Noobie2014, 2021-09-06:
I used a circuit goofing around and it locks up because the NAND gate that triggers the clr pin doesn't get turn back on when its inputs from the outputs of the counter aren't reset to 0.

MarcinOrlowski, 2021-09-07:
@BFH-ktt1 can you take a look? You touched these components recently?

BFH-ktt1, 2021-09-07:
The 74163 has an synchronous clear, the 74161 has an asynchronous clear. Meaning that the 74163 will only reset the outputs when the clear is low and there is a positive edge on the clock input. Hence it is a correct behavior.

maehne, 2021-09-07:
@BFH-ktt1: So this issue can be closed?
@Noobie2014: If you disagree, please attach a ZIP file with the circuit showing the issue and explain observed and expected behaviour.

Noobie2014, 2021-09-07:
yeah i understand the clock on its next rise will reset but it never gets detected because my nand hook up brings the clr low and it doeasn't seem to propogate to the clear part?

BFH-ktt1, 2021-09-08:
There are some small problems in your circuit:

1) At simulation reset, the outputs of your three 74163's are undefined (blue lines), hence you should do a clear at startup.

2) The pull-up on the clear line is redundant, as the nand-gate connected to ```HBackPorch``` is push-pull.

3) The clear functionality of the 74163 is working correctly as the right-most 74163 is correctly resetted (as it has the clock connected to the clock input). The others do not reset correctly as you use a gated-clock connection where the ```TC/RCO``` from one 74163 is connected to the clock input of the next. 

Please try the following:

1) Connect all the clock inputs of the three 74163 to the clock.

2) Connect the ```TC/RCO``` from the right-most 74163 to the ```EnP``` input of the middle 74163.

3) Connect with an and-gate the ```TC/RCO``` from the right-most 74163 and the middle 74163 to the ```EnP``` of the left 74163.

Closing this issue as it is not related to an "incorrect" behavior of the 74163.

closed 2021-09-08 by BFH-ktt1

Noobie2014, 2021-09-08:
Thank you :) I was trying to use the clock on all clk inputs but i just didn't understand the counters. Is there anyway to get local datasheets in the application instead of hunting around on internet? I like your application its very easy to learn and use.

Noobie2014, 2021-09-09:
Thank you and of course you are correct. I am sorry for the stupid mistake, Take care.

maehne, 2021-09-09:
@Noobie2014 wrote:

> Is there any way to get local datasheets in the application instead of hunting around on internet?

This is a documentation issue. Unfortunately, the TTL library is so far not at all documented. If it were the case, we could probably reference the datasheets. Pull requests are welcome!