logisim-evolution/logisim-evolution #1067: Update repo/org configs

MarcinOrlowski, 2021-09-09:
@BFH-ktt1 can you please update both organization and repo config to use consistent graphics:

* repo settings -> Options -> Social preview (upload 1280x640 rendition of main logo)
* organization profile icon (use one of hi-res PNGs images from set of [in-app icons](..src/main/resources/resources/logisim/img/)).

Attached 1280x640 px rendition for repo Social preview -> [lse-1280x640.zip](https://github.com/logisim-evolution/logisim-evolution/files/7135276/lse-1280x640.zip)

BFH-ktt1, 2021-09-10:
Not yet, as long as the branding and icon discussion has not been resolved and voted on. I do not want to make a decision on this topic without the support of all.

MarcinOrlowski, 2021-09-11:
That's fine but can we proceed with the design discussion then and either decide anything or at least say that "nostalgia" is B/S and not an valid argument?

BFH-ktt1, 2021-09-11:
I would suggest to rename the issue to something like ```Update repo icon and preview update descission``` and pin it.
Than ask the community to do a :+1:  or :-1:

MarcinOrlowski, 2022-08-05:
@BFH-ktt1 can we have all the GH icons updated to match project icons for consistency?

MarcinOrlowski, 2022-10-03:
@BFH-ktt1 ping again

BFH-ktt1, 2022-10-04:
@MarcinOrlowski : Sorry, forgot about this one, done.