logisim-evolution/logisim-evolution #1069: Simplifying LineBuffer format placeholder syntax

MarcinOrlowski, 2021-09-10:
Currently LineBuffer format placeholder syntax uses two opening and two closing braces (i.e. `{{foo}}`). I am evaluating if we can safely drop one pair just to have `{foo}`. As we use it mainly for HDL, is there any HDL syntax I miss that could make this tricky and confusing for parser?

BFH-ktt1, 2021-09-10:
As far as I know the ```{``` and ```}``` are not part of either the Verilog and the VHDL language. However, maybe we should take a moment and think if it makes sense to drop the double ```{{```, as I think it makes sense. Also for the readability it is more clear as a single ```{```. And in many editors you can define coloring macros based on the double which might be less obvious for the single.

davidhutchens, 2021-09-10:
On page 66 of The Verilog Golden Reference Guide by DOULOS "{}" is defined as concatenation and replication. I have used it for bit string concatenations and replications. Notice that it is even possible to have two together "{{" or "}}" as the spaces are not required. I'm not sure how this should be reflected here. I just wanted to make everyone aware.

[image]

MarcinOrlowski, 2021-09-11:
Thanks for the feedback. Double brackets stay with this engine then.

BFH-ktt1, 2021-09-11:
@MarcinOrlowski , can we close the issue than?

closed 2021-09-11 by MarcinOrlowski