logisim-evolution/logisim-evolution #1073: Issues with nightly builds

MarcinOrlowski, 2021-09-10:
Nightly builder seem to not build as expected but I am unable to debug without additional information. @BFH-ktt1 can we have debug logs enabled by going to repo `Settings` -> `Secrets` and adding `ACTIONS_STEP_DEBUG` with value `TRUE`.

BFH-ktt1, 2021-09-10:
Done

BFH-ktt1, 2021-09-13:
@MarcinOrlowski : Is there some progress on this one, I see that the latest nightly builds are skipped

MarcinOrlowski, 2021-09-13:
No, I am ATM unable to tell why this happens. I did some research on the subject (people experience imprecise schedule but rather minutes, rarely 1 hour). I also went thru the docs but I failed to find why nightly get it skipped completely.  I think I will restore previous behavior where we had nighly building after each merge. It's not the best solution but I got no better idea ATM.

BFH-ktt1, 2021-09-13:
Now the windows-build fails.....

MarcinOrlowski, 2021-09-13:
Seems it does not like current version string which is probably because of `-dev` suffix. Fixing

davidhutchens, 2021-09-13:
It claims to have worked with a green check after @MarcinOrlowski fixed it. But no packages were actually found and made available. The problem is that it is looking for "3.7.0-dev" rather than "3.7.0dev".

Note that the Windows version will now not have the "dev" at all.

MarcinOrlowski, 2021-09-13:
Indeed, I already noticed that dash is not present i jar name but apparently manged to not fix it. Will do one more take

MarcinOrlowski, 2021-09-28:
Closing, as I believe we have all issues addressed. 

BTW: Github now warns: 

> macOS-latest workflows will use macOS-11 soon. For more details, see https://github.com/actions/virtual-environments/issues/4060

is there anything to bother? @davidhutchens @maehne

closed 2021-09-28 by MarcinOrlowski

davidhutchens, 2021-09-28:
I think we are actually targeting macOS-10.15, so I don't think it will automatically switch us. Ether way, I don't think it is a problem for now.