logisim-evolution/logisim-evolution #1101: IllegalArgumentExceptoin while starting LSe

MarcinOrlowski, 2021-09-14:
### Discussed in https://github.com/logisim-evolution/logisim-evolution/discussions/1095

<div type='discussions-op-text'>

<sup>Originally posted by **marpol1992** September 14, 2021</sup>
I have a problem starting the application version 3.6.0. I have jdk-16.0.2 and jre1.8.0_301 installed, but it returns an error at startup "Failed to launch JVM" </div>

MarcinOrlowski, 2021-09-15:
For a record, this affects only users with system locale set to Polish.

BFH-ktt1, 2021-09-15:
I think that we should maybe plan the roll-out of v3.7.0 soon, and not to try to back port it in v3.6.0, What do you think?

MarcinOrlowski, 2021-09-15:
I am fine with 3.7.0 release. So let's set release date, freeze date and then let's go with [procedure](https://github.com/logisim-evolution/logisim-evolution/discussions/1004).

* #1004

MarcinOrlowski, 2021-09-15:
Closing ticket as fixed.

closed 2021-09-15 by MarcinOrlowski