logisim-evolution/logisim-evolution #1108: Intruducing 64-bits for the components introduced severe bugs in the HDL-generator

BFH-ktt1, 2021-09-15:
Currently I am trying to get the HDL.generator more intelligent, and I discovered several bugs in the HDL-generator related to the extention of the nrOfBits from 32->64. As all HDL-generators use generics, Integers are used parameter maps for VHDL (only 32 bits), for Verilog I have still to check the documentation concerning the parameter maps.
However, many components will not work correctly using more than 32-bits on the hardware, as an example:
The counter has a "Max value", this will be truncated to an integer.....
The complete HDL-generator should be extended to account for the 64-bits.
Staying below 33-bits will not cause any problems.

BFH-ktt1, 2021-09-15:
The same holds for the invering "bubbles" at the input of the gates

R3dst0ne, 2021-09-15:
This was one of my main struggles while working on the new Random generator. I pause my work on it until most changes to the HDL generation get introduced.

BFH-ktt1, 2021-09-16:
PR #1103 fixes these issues

BFH-ktt1, 2021-09-23:
Closing as fixed by #1103

closed 2021-09-23 by BFH-ktt1