logisim-evolution/logisim-evolution #1121: Line 17: Invalid variable name '|'.

GlowingUmbreon, 2021-09-24:
When trying to import a table I get the following error. 
[image]
This seems to happen on every table you import even ones you exported, here is a table that will cause the error:
```
# Truth table
# Generated from circuit test2
# Exported on Fri Sep 24 15:44:44 BST 2021

# Hints and Notes on Formatting:
# * You can edit this file then import it back into Logisim!
# * Anything after a '#' is a comment and will be ignored.
# * Blank lines and separator lines (e.g., ~~~~~~) are ignored.
# * Keep column names simple (no spaces, punctuation, etc.)
# * 'Name[N..0]' indicates an N+1 bit variable, whereas
#   'Name' by itself indicates a 1-bit variable.
# * You can use 'x' or '-' to indicate "don't care" for both
#   input and output bits.
# * You can use binary (e.g., '10100011xxxx') notation or
#   or hex (e.g., 'C3x'). Logisim will figure out which is which.

A B | Z Y
~~~~~~~~~
0 0 | 1 1
0 1 | 1 0
1 0 | 0 1
1 1 | 0 0
```

MarcinOrlowski, 2021-09-24:
Can you please attach the raw file? You may need to ZIP it first though

GlowingUmbreon, 2021-09-24:
[nom.txt](https://github.com/logisim-evolution/logisim-evolution/files/7227842/nom.txt)

yaybee, 2021-09-25:
I just opened a new issue for vector tests, but I notice as well, that the Hints and Notes on test vectors are not correct.
Next to the character | also ~ is not allowed as ignored input and - can not be used for "don't care".

BFH-ktt1, 2021-09-26:
I confirm the bug, the csv-format does the correct thing, the .txt file doesn't.

BFH-ktt1, 2021-09-26:
Fixed in PR #1126, please reopen if the problem persists

closed 2021-09-26 by BFH-ktt1