logisim-evolution/logisim-evolution #1123: Cannot use "don't care" character "x" on inputs of the vector test function

yaybee, 2021-09-25:
Cannot use "don't care" character "x" on inputs of the vector test function

The vector test function on logical gates behaves strangely with "don't care" character "x" on inputs.
The documentation says:
	For hex, octal, and binary values, a digit of 'x' specifies four, three, or one "don't care" bits.

For me this means: x can be either 0 or 1, but the vector testing is using "undefined" at least for one bit inputs.
This will give an error signal on output wire of a gate and results in a failing test and is therefor not usable on input ports.

See example with a NAND gate test vector:
[image]

Test outcome:
[image]

Circuit state:
[image]

When using "don't care" character on inputs the vector testing should do tests on all input combination like the function Analyse Circuit does.
The results of these tests can than be compared with the given test vectors to see if tests fail or pass
Now "don't care" input and outputs can be masked in the compare process to allow a zero and an one.

Do we have a bug here of are "don't care" characters not intended to be used on gate inputs?

BFH-ktt1, 2021-09-26:
The test-vector feature does not support don't cares for the input pins for the moment

maehne, 2021-09-29:
When this issue gets worked on, we may also check whether using `x` as "don't care" in the data format doesn't cause additional problems when it also gets used as part of the "0x" prefix for marking hex values. Maybe, it would be better to use in the future rather `-` instead. (I am aware that the use of "x" for "don't care" is common in the context of Verilog due it uses a 4-value logic type, which leaves no space for dedicated "don't care".)

btb, 2022-06-27:
This does not look like a bug to me, I think the "x" feature is working as intended - if one of the inputs of a NAND gate is 1, then the output depends on what the other gate is. Therefore the gate DOES care what its other input is, so your test vector file as constructed does not make sense.

btb, 2022-06-27:
This test vector passes for the given circuit.
```
# test vector for NAND gate
A B Y
0 x 1
x 0 1
1 1 0
```

maehne, 2026-09-07:
@hewzhew: Thanks for your analysis and tests of the current behaviour. I agree that improving the test vector documentation as outlined is a good way to move forward to resolve this old open issue. If we receive test cases where this `U` propagation into the combinatoric circuit does not properly yield the specified result, we can reconsider whether actually all combinations satisfying the input constraint need to applied sequentially to check whether their outputs agree to the specified case.

closed 2026-09-29 by BFH-ktt1