logisim-evolution/logisim-evolution #1139: HDL-generated code identation needs attention

BFH-ktt1, 2021-09-27:
Since the introduction of the LineBuffer the indentation of some of the generated HDL-code got kind-of screwed up, example:
```
--==============================================================================
--== Logisim-evolution goes FPGA automatic generated VHDL code                ==
--== https://github.com/logisim-evolution/                                    ==
--==                                                                          ==
--==                                                                          ==
--== Project   : leguantest                                                   ==
--== Component : LogisimCounter                                               ==
--==                                                                          ==
--==============================================================================

ARCHITECTURE PlatformIndependent OF LogisimCounter IS 

--------------------------------------------------------------------------------
-- Here all used signals are defined                                          --
--------------------------------------------------------------------------------
   SIGNAL s_clock                            : std_logic;
   SIGNAL s_real_enable                      : std_logic;
   SIGNAL s_next_counter_value               : std_logic_vector( (width-1) DOWNTO 0 );
   SIGNAL s_carry                            : std_logic;
   SIGNAL s_counter_value                    : std_logic_vector( (width-1) DOWNTO 0 );

BEGIN
   --------------------------------------------------------------------------------
   -- Functionality of the counter:                                              --
   -- __Load_Count_|_mode                                                        --
   -- ____0____0___|_halt                                                        --
   -- ____0____1___|_count_up_(default)                                          --
   -- ____1____0___|load                                                         --
   -- ____1____1___|_count_down                                                  --
   --------------------------------------------------------------------------------
   CompareOut   <= s_carry;
CountValue   <= s_counter_value;

s_clock      <= clock WHEN InvertClock = 0 ELSE NOT(clock);
make_carry : PROCESS( Up_n_Down,
                      s_counter_value )
BEGIN
   IF (Up_n_Down = '0') THEN
      IF (s_counter_value = std_logic_vector(to_unsigned(0,width))) THEN
         s_carry <= '1';
      ELSE
         s_carry <= '0';
      END IF; -- Down counting
   ELSE
      IF (s_counter_value = max_val) THEN
         s_carry <= '1';
      ELSE
         s_carry <= '0';
      END IF; -- Up counting
   END IF;
END PROCESS make_carry;

s_real_enable <= '0' WHEN (load = '0' AND enable = '0') -- Counter disabled
                       OR (mode = 1 AND s_carry = '1' AND load = '0') -- Stay at value situation
                     ELSE tick;

make_next_value : PROCESS( load , Up_n_Down , s_counter_value ,
                           LoadData , s_carry )
   VARIABLE v_downcount : std_logic;
BEGIN
   v_downcount := NOT(Up_n_Down);
   IF ((load = '1') OR -- load condition
       (mode = 3 AND s_carry = '1')    -- Wrap load condition
      ) THEN s_next_counter_value <= LoadData;
   ELSE
      CASE (mode) IS
         WHEN  0    => IF (s_carry = '1') THEN
                          IF (v_downcount = '1') THEN
                             s_next_counter_value <= max_val;
                          ELSE
                             s_next_counter_value <= (OTHERS => '0');
                          END IF;
                       ELSE
                          IF (v_downcount = '1') THEN
                             s_next_counter_value <= std_logic_vector(unsigned(s_counter_value) - 1);
                          ELSE
                             s_next_counter_value <= std_logic_vector(unsigned(s_counter_value) + 1);
                          END IF;
                       END IF;
        WHEN OTHERS => IF (v_downcount = '1') THEN
                           s_next_counter_value <= std_logic_vector(unsigned(s_counter_value) - 1);
                       ELSE
                           s_next_counter_value <= std_logic_vector(unsigned(s_counter_value) + 1);
                       END IF;
      END CASE;
   END IF;
END PROCESS make_next_value;

make_flops : PROCESS( s_clock , s_real_enable , clear , s_next_counter_value )
BEGIN
   IF (clear = '1') THEN s_counter_value <= (OTHERS => '0');
   ELSIF (rising_edge(s_clock)) THEN
      IF (s_real_enable = '1') THEN s_counter_value <= s_next_counter_value;
      END IF;
   END IF;
END PROCESS make_flops;

END PlatformIndependent;
```

BFH-ktt1, 2021-09-28:
I could think of a method, for example ```setIndentLevel(int level)``` that is applied to all ```.add( )``` methods afterwards.
also a method like ```increaseIdentLevel()``` and ```decreaseIdentLevel()``` might be usefull. This of course means that we have to go through all HDL-generators, but this is a one-time effort. What do you think?

BFH-ktt1, 2021-09-28:
Also a handle like ```addAllign("SIGNAL s_clock : std_logic;" , ':');``` could be very usefull for HDL for the Generics, Ports and Signals. Where we can align automatically a block of signals by the one with the largest name and the separation symbol.

maehne, 2021-09-29:
I don't think that a method for setting the absolute indent level is much useful, as it might screw up indentation easily if it gets overlooked at later refactoring passes. It would be better to explicitly increase/decrease indent levels. Maybe, this could be automatically done to some degree by code, which anyway outputs HDL code describing a scope (e.g., entities, architectures, processes).

Additional methods trying to ensure further alignment, e.g., in signal declarations may proove fragile too, e.g., in the case signal names or type names get long, which may result in overly long code lines.

MarcinOrlowski, 2021-09-29:
I think that while formatting helpers may be useful to have, fixing indentation by adding methods like `setIndentLevel()` is not the addressing the issue at its roots. It would be much better to have indentation handled automatically based of the context or role/type of generated block. I do mean that if we would be able to tell which method generates what (like "this is header", "this is loop", "this is xxx") then it could be nicely automated.

BFH-ktt1, 2021-09-29:
I found that many of these issues are just cases where .getwithIndent() should be replaced by .get(). It is really not an issue of LineBuffer, nor of the changes introduced by LineBuffer, but more, I think, about the (my) understanding from .getwithIndent(x) where x represents the number of indent levels and not the number of spaces :-). I'm on it and close this issue, as it will be resolved soon :-)

closed 2021-09-29 by BFH-ktt1