logisim-evolution/logisim-evolution #1140: Crash on starup with Polish localization

Paliak, 2021-09-27:
Basically #477 but seems like it's broken again in 3.6.1 release.

Using the windows installer results in failed to launch jvm popup when logisim-evolution.exe is ran.

Running the jar with openjdk downloaded from https://jdk.java.net/16/ gives:

```
F:\Downloads\openjdk-16.0.2_windows-x64_bin\jdk-16.0.2\bin>java -jar "C:\Program Files\logisim-evolution\app\logisim-evolution-3.6.1-all.jar"
Exception in thread "main" java.lang.ExceptionInInitializerError
        at com.cburch.logisim.prefs.AppPreferences$LocalePreference.<init>(AppPreferences.java:89)
        at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:477)
        at com.cburch.logisim.Main.main(Main.java:65)
Caused by: java.lang.IllegalArgumentException: Illegal pattern character 'r'
        at java.base/java.text.SimpleDateFormat.compile(SimpleDateFormat.java:846)
        at java.base/java.text.SimpleDateFormat.initialize(SimpleDateFormat.java:654)
        at java.base/java.text.SimpleDateFormat.<init>(SimpleDateFormat.java:625)
        at java.base/java.text.SimpleDateFormat.<init>(SimpleDateFormat.java:600)
        at com.cburch.logisim.util.LocaleManager.<clinit>(LocaleManager.java:271)
        ... 3 more

F:\Downloads\openjdk-16.0.2_windows-x64_bin\jdk-16.0.2\bin>java --version
openjdk 16.0.2 2021-07-20
OpenJDK Runtime Environment (build 16.0.2+7-67)
OpenJDK 64-Bit Server VM (build 16.0.2+7-67, mixed mode, sharing)
```

The fix as mentioned in #477 is to change the "dateFormat" parameter in logisim-evolution-3.6.1-all.jar\resources\logisim\strings\util\util_pl.properties from:
`
dateFormat = dd.MM.rrrr HH:mm:ss`
to:
`dateFormat = dd.MM.yyyy HH:mm:ss`

After saving the file and updating the archive using 7z, starting using both the seperate openjdk and the included logisim-evolution.exe seems to work fine.

MarcinOrlowski, 2021-09-27:
Most likely it is because v3.6.1 is based on 3.6.0 code which still have this problem and @BFH-ktt1 only included specific fix for FPGA board he had problems with. Localization issue was fixed `develop` branch only and we did not back-port it to `master` since then. @Paliak how crucial is this issue for you? Can you use nightly instead of official stable build?

Paliak, 2021-09-28:
Ahh. Took me a good bit of debugging to fix it so i thought i'd post my findings and right before creating a new issue i found #447. Saw that it was from January so i assumed it'd be implemented in master by now but didn't actually check (debugging was a bit of a headache, took me a while to figure out that the issue was with localization), sorry.

Not a huge deal for me. The modified jar works fine for my needs and yes i could use the nightly builds.

Would it be possible to add a warning somewhere in the README or maybe disable the Polish localization until the solution is implemented ?

MarcinOrlowski, 2021-09-28:
We either release 3.7.0 (which includes this fix) or if for any reason 3.7.0 will still need some time to bake, 3.6.2 will be out fixing that.

MarcinOrlowski, 2021-09-28:
I filled #1150 to have this addressed either way.

closed 2021-09-28 by Paliak