logisim-evolution/logisim-evolution #1142: Please don't be angry but I think the 74163 parallel load is broke.

Noobie2014, 2021-09-28:
I last sent an erroneous issue and am sorry for the mistake but this time i think i found one. I am sending this circuit for you to debug the issue. I have a 74161 and a 74163 wired the same and I
[soloCntr.zip](https://github.com/logisim-evolution/logisim-evolution/files/7241275/soloCntr.zip)
 can get the 74161 to load a parallel input but for the life of me I can't get the 74163 to load one. The chips are identical all except the clear input for one is asynchronous and the other synchronous.

BFH-ktt1, 2021-09-28:
Thanks for your Issue!

MarcinOrlowski, 2021-09-28:
> I last sent an erroneous issue and am sorry for the mistake but this time i think i found one. 

No worries @Noobie2014. This is nothing unusual. Thanks for spending the time to file the ticket (regardless of the final outcome).

MarcinOrlowski, 2021-09-28:
Closing as fixed (please grab nightly and try). Please reopen if needed.

PR #1143 (2021-09-28): Fixed bug #1142
Closes #1142

* #1142
closed 2021-09-28 by MarcinOrlowski