logisim-evolution/logisim-evolution #1144: HDL-generators contain still a lot of hard coded Strings that are not localized

BFH-ktt1, 2021-09-28:
In many of the HDL-generators errors and warnings are still hard coded with Englisch strings. These should be localized.

MarcinOrlowski, 2021-09-28:
I suggest (which is what I am personally doing) to mark all these places with

    // FIXME: hardcoded string

comment, so when time come, it will be much easier to deal with it.

maehne, 2026-08-21:
Thanks @hewzhew for your clean ups! Am I right that you did not yet open a PR for this fifth slice?

To be able to track the completion state of this issue, a bullet list showing the strings that are hard-coded and still need clean up would be helpful.

hewzhew, 2026-08-21:
Thanks for checking, @maehne. Yes, that is correct: the fifth slice has only been scoped in the comment above; I have not yet implemented it or opened a PR. I will take it next.

I re-audited current `main`. For tracking purposes, the completed slices are:

- #2824: the two component-specific missing/gated clock warnings in `AbstractHdlGeneratorFactory`;
- #2825: the malformed board-file errors in `BoardReaderClass`;
- #2835: the empty entity/architecture diagnostics in `Hdl`; and
- #2849: the two invalid connection-point diagnostics in `Hdl.getNetMap()` and `CircuitHdlGeneratorFactory.getSignalMap()`.

The following user-visible `Reporter` messages in the HDL-generation and adjacent FPGA download path are still hard-coded:

- **Proposed fifth slice:** the three unconnected-output grounding warnings in `CircuitHdlGeneratorFactory.getSignalMap()` (one-bit output, wholly unconnected output bus, and one unconnected bit of a partially connected output bus).
- **Other circuit-generation diagnostics in `CircuitHdlGeneratorFactory`:** missing component HDL generator; missing subcircuit HDL generator; disconnected clock component being skipped; missing clock net; missing I/O pin mapping; and failure to find a subcircuit port end index (the latter currently occurs in three port-direction paths with the same wording).
- **Shared/top-level generators:** the missing parameter diagnostic in `AbstractHdlGeneratorFactory` and the zero-pin component diagnostic in `ToplevelHdlGeneratorFactory`.
- **Component generators:** the floating Bit Extender input diagnostic and the unsupported-Verilog diagnostic in `BinToBcdHdlGeneratorFactory`.
- **Generation/download orchestration:** unknown download target and missing top-level sheet in `Download`; and in `DownloadBase`, missing circuit/board information, the board-content information lines, directory creation/cleanup/check/removal failures, and the null top-level HDL generator diagnostic.

I limited this inventory to messages delivered through `Reporter` during HDL generation/download. There are also adjacent hard-coded GUI titles in the FPGA board/mapping dialogs and hard-coded VHDL-simulator startup dialog text, but I did not count those as HDL-generator diagnostics; they can be tracked as separate UI/simulator cleanups if desired. Generated HDL syntax/identifiers and debug-only logger text are deliberately excluded.

After the fifth slice, I suggest keeping the remaining work reviewable as topical batches: circuit-generation internals, shared/component generators, and generation/download orchestration.

maehne, 2026-08-21:
@hewzhew: Thanks for compiling the list, which I converted to a checkbox list for keeping track. You can proceed as suggested.