logisim-evolution/logisim-evolution #1158: Compile and Simulation broken

davidhutchens, 2021-09-29:
The current version of develop is broken. It does not compile because line 77 of src/main/java/com/cburch/logisim/gui/main/SimulationTreeModel.java is null when it should be 0. But changing that leads to a compiled version that will not turn on the clock to run the simulation.

BFH-ktt1, 2021-09-29:
@davidhutchens , could you give an example, as I can not reproduce on my side, the attached circuit is working perfectly for me with the last pushes:
[leguantest.zip](https://github.com/logisim-evolution/logisim-evolution/files/7254631/leguantest.zip)

MarcinOrlowski, 2021-09-29:
Works for me. Same HEAD:

```
$ git log --all --decorate --oneline --graph | head -5
*   626b0cf85 (HEAD -> develop, upstream/develop) Merge pull request #1154 from MarcinOrlowski/instanceof-pm2
|\
| * 9cf41f5bf Refactored use of instanceof
| * a3710b386 Refactore use of instanceof
* |   433ea2cce Merge pull request #1155 from MarcinOrlowski/arguments-fix
```

Builds fine:

```
./gradlew build -x checkstyleMain -x checkstyleTest
Starting a Gradle Daemon (subsequent builds will be faster)
Unable to watch the file system for changes. The inotify instance limit is too low. See https://docs.gradle.org/7.0.2/userguide/gradle_daemon.html#sec:inotify_instances_limit for more details.

BUILD SUCCESSFUL in 23s
13 actionable tasks: 12 executed, 1 up-to-date
Gradle was unable to watch the file system for changes. The inotify instance limit is too low. See https://docs.gradle.org/7.0.2/userguide/gradle_daemon.html#sec:inotify_instances_limit for more details.
```

MarcinOrlowski, 2021-09-29:
Also note, that our Github builders also work fine.

davidhutchens, 2021-09-29:
Curious. I'll investigate.

davidhutchens, 2021-09-29:
I checked the file on github. Here is the method that does not compile for me:

` @Override
  public int getChildCount(Object parent) {
    return (parent instanceof TreeNode node)
           ? node.getChildCount()
           : null;
  }
`

Clearly null is not the right type for a method returning int, which is what my build complains about.

Investigation shows this does compile with Java 16 and fails to compile in Java 17!

davidhutchens, 2021-09-29:
I have simple circuits that work and a complex one that used to work and now fails run in simulation. It may take some time to figure out what part is different.

davidhutchens, 2021-09-29:
My complex circuit has lots of subcircuits. I haven't come up with a simple example yet as I just noticed that there are values presented to the outputs of subcircuits that are not being passed to the outer circuit and inputs of subcircuits are not getting values from the outer circuit. The other side is just undefined.

I'll see if I can get a simple example, but I thought that much might trigger something.

davidhutchens, 2021-09-29:
Here is an example of a subcircuit that outputs undefined rather than the value it computes.
[SimpleSubCircuit.circ.zip](https://github.com/logisim-evolution/logisim-evolution/files/7254932/SimpleSubCircuit.circ.zip)

MarcinOrlowski, 2021-09-29:
Are these two things related? I already fixed this `null` thing and it was a bug. If you correct this, is simulation misbehaving still?

PS: It's interesting that Checkstyle failed to catch that too.

davidhutchens, 2021-09-29:
Fixing the `null` issue does not help with the simulation problem.

The simulation problem is flaky, so it is not easy to get a simple example. I'm trying.

MarcinOrlowski, 2021-09-29:
Can you please create separate ticket for simulation issues?

davidhutchens, 2021-09-30:
Simulation problem moved to #1160.

This issue should be close when #1159 is merged

MarcinOrlowski, 2021-09-30:
Closing as fixed.

PR #1159 (2021-09-30): Fixed wrong return value in getChildCount()
Fixes #1158

* #1158 1158
closed 2021-09-30 by MarcinOrlowski