logisim-evolution/logisim-evolution #1160: Simulation Problem with current develop branch

davidhutchens, 2021-09-29:
Here is a sample circuit that when loaded in the current develop branch, shows the output of the subcircuit as undefined, but if you look in the subcircuit, the output is zero. It works correctly in earlier builds.

[SimulationProblem1.circ.zip](https://github.com/logisim-evolution/logisim-evolution/files/7255189/SimulationProblem1.circ.zip)

MarcinOrlowski, 2021-09-29:
>  It works correctly in earlier builds.

David, will you be able to narrow the commit that broke it?

davidhutchens, 2021-09-29:
Likely, yes. I had other things to do for a while. I'll look at it later.

MarcinOrlowski, 2021-09-30:
Great, thanks. You are aware of bisection approach that can save you some time and effort there?

davidhutchens, 2021-09-30:
Aware of a bisection approach? Yeah. I may have been teaching binary searches before you were born :-). Anyway, commit da88fe975531d235c0f09362c3c071a01a545148 is good. 19f3ed03998659f7c2e1c6482a991c9f46320ab1 doesn't work. d9cd0fcf939f223686f8e11a005fa2c028aae0be is in the middle of them, but doesn't compile.

MarcinOrlowski, 2021-09-30:
Heh, that's possible :) But I thought compilation issue is one thing (which is now fixed) and simulation problems is another, no? As #1159 is merged project should compile just fine (unless I miss something?) but you said `d9cd0fc is in the middle of them, but doesn't compile.` however the `null` bug was introduced in `a3710b38683eb3dcf67d867223c0450820202462`:

```bash
$ git show a3710b3868 --stat
commit a3710b38683eb3dcf67d867223c0450820202462
Date:   Wed Sep 29 11:32:33 2021 +0200
```

which is **later** commit than `d9cd0fc`:

```bash
$ git show d9cd0fc --stat
commit d9cd0fcf939f223686f8e11a005fa2c028aae0be
Date:   Tue Sep 28 14:45:40 2021 +0200
```

What do I misunderstand?

davidhutchens, 2021-09-30:
I was testing in Java 16, so the null bug was not the cause of the failure in d9cd0fcf939f223686f8e11a005fa2c028aae0be. It fails with:
`logisim-evolution/src/main/java/com/cburch/logisim/fpga/hdlgenerator/InlinedHdlGeneratorFactory.java:19: error: InlinedHdlGeneratorFactory is not abstract and does not override abstract method generateAllHDLDescriptions(Set<String>,String,List<String>) in HdlGeneratorFactory
public class InlinedHdlGeneratorFactory implements HdlGeneratorFactory {
       ^
`

MarcinOrlowski, 2021-09-30:
Hm, [but for me it does](https://github.com/logisim-evolution/logisim-evolution/blob/develop/src/main/java/com/cburch/logisim/fpga/hdlgenerator/InlinedHdlGeneratorFactory.java#L22)? How recent is your develop?l

davidhutchens, 2021-09-30:
I'm confused by your comment. The three commits I listed above were in a row in the history. The one "in the middle" did  not compile. The compile problems were fixed in the following commit (third of the series), but that one fails to run properly. The bug for this issue was introduced in either d9cd0fcf939f223686f8e11a005fa2c028aae0be or 19f3ed03998659f7c2e1c6482a991c9f46320ab1.

My develop branch was up-to-date as of yesterday. The only commit with the failure to compile because of the override issues is d9cd0fcf939f223686f8e11a005fa2c028aae0be. My current HEAD of develop compiles (well, in Java 16).

BFH-ktt1, 2021-10-01:
@davidhutchens and @MarcinOrlowski : What is the status of this investigation?

davidhutchens, 2021-10-01:
@BFH-ktt1 and @MarcinOrlowski : I have tried to find the cause by examining the commits that introduced it. I have not yet been successful. I am not very familiar with the simulation engine.

But I can more reliably see the problem. I added a very simple example to the initial comment above and attempted to describe what I see. Can either of you reproduce that, or at least see strange behavior? It still occurs in today's develop branch.

I view this as a critical bug since I believe it will cause failures on any .circ file that contains sub-circuits. Please do not release 3.7.0 until this is resolved.

MarcinOrlowski, 2021-10-01:
@davidhutchens which commit is the one that makes LSe expose this issue?

MarcinOrlowski, 2021-10-01:
> The bug for this issue was introduced in either d9cd0fc or 19f3ed0.

Both are mine. Lemme see...

MarcinOrlowski, 2021-10-01:
I fixed compilation issues for d9cd0fcf939f223686f8e11a005fa2c028aae0be and it exposes the issue so whatever causing this was introduced here.

BFH-ktt1, 2021-10-01:
@davidhutchens , @MarcinOrlowski : Found it, it was in [d9cd0fc](https://github.com/logisim-evolution/logisim-evolution/commit/d9cd0fcf939f223686f8e11a005fa2c028aae0be) in ```CircuitState.java```:
[image]

Made a PR

closed 2021-10-01 by davidhutchens