logisim-evolution/logisim-evolution #1167: Funny effect with LineBuffer that I cannot explain in the moment

BFH-ktt1, 2021-09-30:
Line 334-342 in CircuitHDLGeneratorFacotry.java:

```
    for (var i = 0; i < theNetList.getNumberOfInputPorts(); i++) {
      final var myInput = theNetList.getInputPin(i);
      final var pinName = CorrectLabel.getCorrectLabel(myInput.getComponent().getAttributeSet().getValue(StdAttr.LABEL)); 
      wires.putAll(getSignalMap(pinName, myInput, 0, theNetList));
    }
    if (!wires.isEmpty()) {
      contents.empty().addRemarkBlock("Here all input connections are defined");
      Hdl.addAllWiresSorted(contents, wires);
    }
```

```getSignalMap``` is from line 565 in the same file (will not copy it here)

```Hdl.addAllWiresSorted(contents, wires);``` is define by:

```
  public static void addAllWiresSorted(LineBuffer contents, Map<String, String> wires) {
    var maxNameLength = 0;
    for (var wire : wires.keySet())
      maxNameLength = Math.max(maxNameLength, wire.length());
    final var sortedWires = new TreeSet<String>(wires.keySet());
    for (var wire : sortedWires) 
      contents.add("   {{assign}}{{1}}{{2}} {{=}} {{3}};", wire, " ".repeat(maxNameLength - wire.length()), wires.get(wire));
    wires.clear();
  }
```

And in case of buses and wires this generates:

```
   --------------------------------------------------------------------------------
   -- Here all input connections are defined                                     --
   --------------------------------------------------------------------------------
      s_LOGISIM_BUS_8(3 DOWNTO 0)  <=  D;
      s_LOGISIM_NET_0                      <=  R;
      s_LOGISIM_NET_1                      <=  CLK;
      s_LOGISIM_NET_2                      <=  WE;
```

hence in some sense the parameterized variables are taking into account when calculating the max. string size (which would account for the extra spaces). I did not look into all details, but find it a funny effect, as for all others the function ```addAllWiresSorted``` does the correct job....

MarcinOrlowski, 2021-09-30:
I'd set a breakpoint after `wires.putAll()` and inspect the `wires`

BFH-ktt1, 2021-10-02:
Found the problem and fixed, closing

commit a590b892 (2021-10-07): Fixed #1167 and more VHDL cleanups
closed 2021-10-02 by BFH-ktt1