logisim-evolution/logisim-evolution #1182: Classes rename to follow naming convention

MarcinOrlowski, 2021-10-02:
The  #1181 PR renames most classes to follow naming convention, yet a few classes are still unaltered due to lack of idea what to rename them to. These are:

* ZOrder
* RV32im_M_ExtensionInstructions
* DFlipFlop
* JKFlipFlop
* SRFlipFlop
* TFlipFlop

and multiple Swig related classes:
* J(dialog/input/file)...

I am open to hear if anyone got some ideas.

*  #1181

maehne, 2021-10-04:
I always try to follow a systematic naming conventions in my code. Nevertheless, we have to live with the fact that there are always edge cases, where we cannot fully satisfy them. Therefore, I am fine with the six class names, which you list first.

@BFH-ktt1 may have a better suggestion for `RV32im_M_ExtensionInstructions`.

Regarding the Swing-related classes, we should follow Swing conventions -- even if these are in some points in conflict with our own conventions. Though, we may consider renaming such classes if the make a wrong impression to users that they are official part of Swing.

MarcinOrlowski, 2021-10-04:
> @BFH-ktt1 may have a better suggestion for RV32im_M_ExtensionInstructions.

Can it be say `RV32imExtensionInstructionsM` (as all classes in that package start with `RV32im`). We can however also stick to `RV32imMExtensionInstructions` but that simply looks bad (multiple `m`, two capitals etc)

> if the make a wrong impression to users that they are official part of Swing.

That's actually a good point.

maehne, 2021-10-04:
Well, the `M` stands for something. Maybe, it could be slightly expanded. Though, the class name is already quite long. Let's wait for a feedback by @BFH-ktt1.

BFH-ktt1, 2021-10-04:
Well in the Risc-5 ISA for the there are several instruction types, all referred to by a letter. I would go for @MarcinOrlowski proposal of naming it ```RV32imExtensionInstructionsM```, and keep this naming convention consistent over the other instruction types.

BFH-ktt1, 2021-10-04:
B.t.w. the M stands for multiply, but than the name get's enormous: ```RV32imMultiplyExtensionInstructions```

hewzhew, 2026-08-19:
I rechecked this issue against current `main`. The `ZOrder` and flip-flop names do not seem to need revisiting, while the specific `RV32im_M_ExtensionInstructions` rename proposed in the discussion is still pending.

A later class, `RV32im_Zicsr_ExtensionInstructions`, now follows the same underscore pattern. I propose a narrow, behavior-preserving cleanup:

- rename the two internal classes to `RV32imExtensionInstructionsM` and `RV32imExtensionInstructionsZicsr`;
- update their registrations in `RV32imAssembler`;
- leave all other names discussed here unchanged.

I found no reflection lookups, resource keys, serialization references, or other uses of these class names. Compilation and Checkstyle should therefore provide focused validation, followed by the normal test suite. Does this scope still make sense?

maehne, 2026-08-21:
Personally, I am fine with the proposed names `RV32imExtensionInstructionsM` and `RV32imExtensionInstructionsZicsr`, but I leave the decision to @BFH-ktt1, as he wrote that part of Logisim-evolution.