logisim-evolution/logisim-evolution #1189: LineBuffer textblock indenting issue

BFH-ktt1, 2021-10-02:
I observe a strange behavior:

```
public LineBuffer getModuleFunctionality(Netlist TheNetlist, AttributeSet attrs) {
...
      contents.addVhdlKeywords().add("""
          s_tick_next   <= '1' {{when}} s_count_reg = std_logic_vector(to_unsigned(0, {{nrOfCounterBits}})) {{else}} '0';
          s_count_next  <= ({{others}} => '0') {{when}} s_tick_reg /= '0' {{and}} s_tick_reg /= '1' {{else}} -- For simulation only!
                           std_logic_vector(to_unsigned((ReloadValue-1), {{nrOfCounterBits}})) {{when}} s_tick_next = '1' {{else}}
                           std_logic_vector(unsigned(s_count_reg)-1);

          """);
```

than the ```.getWithIndent()``` produces:

```
   s_tick_next   <= '1' WHEN s_count_reg = std_logic_vector(to_unsigned(0, nrOfBits)) ELSE '0';
s_count_next  <= (OTHERS => '0') WHEN s_tick_reg /= '0' AND s_tick_reg /= '1' ELSE -- For simulation only!
                 std_logic_vector(to_unsigned((ReloadValue-1), nrOfBits)) WHEN s_tick_next = '1' ELSE
                 std_logic_vector(unsigned(s_count_reg)-1);
```

hence the first line is correctly indented, but all the following once not. This is consistent over all the text blocks, the first one get's correctly indented, the other lines not.

MarcinOrlowski, 2021-10-02:
Well, this works correctly though not as you expect. The reason is that text block is being added as simgle string so there is also singke string to indent. I will change this to break data into single lines

BFH-ktt1, 2021-10-02:
@MarcinOrlowski : I already implemented the line-splitting in LineBuffer, closing the issue

closed 2021-10-02 by BFH-ktt1

MarcinOrlowski, 2021-10-02:
Have you also created unit tests for this feature? If now, please reopen.