logisim-evolution/logisim-evolution #1192: Bug introduced by #1190 or #1191, subcircuits are not placed in schematic

BFH-ktt1, 2021-10-03:
Very probable it is #1191 that causes the problem. Most of the sub-circuits are not any more placed in the circuit sheets. No error is given and no exception is found. Example for a system that used to work before #1190 is attached.
[Testbench.zip](https://github.com/logisim-evolution/logisim-evolution/files/7272919/Testbench.zip)

MarcinOrlowski, 2021-10-03:
Hm, seems some of my tools are too aggressive. Will check.

davidhutchens, 2021-10-03:
The problem is in commit 4e3e1b973b5121da117af61b0f976dfdd2dd0f36 of #1191.

There are definitely places where your tools may have been too aggressive. For example:

[image]

The original returns file if libName is null, the changed one does not.

However, restoring that logic does not solve this problem. So more searching is required.

davidhutchens, 2021-10-03:
Found it!
[image]

This fails to assign the return value to comp. This fixes the problem. I'll prepare a PR.

BFH-ktt1, 2021-10-04:
@davidhutchens : Thanks a lot for finding the bug. 
@MarcinOrlowski : Could you check that there are no other places where the tools were too agressive introducing bugs that we did not discover yet?

MarcinOrlowski, 2021-10-04:
Definitely I will. Luckily the tool wasn't used any earlier than these two PRs.  

BTW: can we also agree that no reviewer gives his approval (and no PR is merged) unless it is least read? I know these cleanup PRs are boring lecture and it's often cross domain (like when one changes both the HDL generated and code that generates it) and sometimes it might require more than one reviewer to cover the whole scope, but that'd save us some time chasing the bugs later. Also I am thinking if we should not limit the size of the diff for each PR (nor hard-limit, but still a limit) to encourage/enforce smaller PRs. In most cases these PRs can be split into smaller chunks, and one can easily chain-branch from and use `DEPEND_ON` to enforce specified order of merging if needed.

@BFH-ktt1 @maehne @davidhutchens

BFH-ktt1, 2021-10-04:
@MarcinOrlowski : I went through the PR's, but I also depend on the assumption that a person submitting a PR at least tests that no things got broken; hence went also through his own PR's. Small errors can always slip in, that's why there is a review process (4 eyes see more than 2).

maehne, 2021-10-04:
@MarcinOrlowski: As long as PRs stay topical and fix one issue at a time, I don't have any problem if they are long. I always try to read through the whole PRs and places, where a lot of lines are removed/added at once draw my attention. But, it can be hard to grasp the real scope of such larger modifications if there are no helpful comments in the code, commit messages, or in the PR comments. Feel free to split up more your bigger PRs in a chain of smaller PRs. I agree that it should be good practice and is assumed by us that PR submitters review their modifications themselves first -- especially if these were done semi-automatically by some refactoring tool.

maehne, 2021-10-04:
@BFH-ktt1: As PR #1193 got merged, can we close this issue?

MarcinOrlowski, 2021-10-04:
If you read it prior approval, then all is fine, thanks. My point about having smaller PRs is that one (and I felt into this a few times too) is sometimes tempted to do more than minimal scope. For example if I rename one method and that would affect 200 files, then this is still pretty simple to review. But if I additionally add change some logic, or do additional refactoring, then things get complicated.

maehne, 2021-10-04:
Maybe, we should make it practice that larger refactoring PRs get reviewed by two and not only one main developer. Also, it would be nice if a PR message would draw with additional comments attention of reviewers to especially critical sections of the PR that require a more close look. In that case, it may also help to first make a PR with the easy modifications (pure clean ups / fixes to names, etc.), which are unlikely to cause trouble and then a depending PR, which makes optimisations by replacing some existing logic with a new/simpler implementation.

davidhutchens, 2021-10-04:
I agree with many of the comments above. The debugging issue is more the size of the commits than the PRs; but reducing the size of the PRs may naturally reduce the size of the commits as well. The reviewing effort really is the size of the PR.

As for testing, the PR with this issue touched many things in file I/O. Before submitting my PR last night, I made sure to take a complex example and SaveAs to a new file. Then I opened that file to see that things still worked. That way, I tested both reading and writing of .circ files. That was still limited and didn't test other types of files (e.g. .jar files), and we probably should. Part of the problem is that we each use LE in different ways and don't necessarily have a wide set of test files. Maybe we should revisit the long-standing goal of building an extensive automatic test set.

maehne, 2021-10-04:
Maybe as a first step, we could start collecting a number of benchmark designs of different complexity, which could be used for manual testing. In a second step, some of them could be extended in some way to produce a kind of simulation trace, which depend on some reference stimuli, which could be compared to a known golden reference trace file. One way this could work, would be implementing the stimuli generation/signal monitoring as a VHDL (test bench) entity. Besides internal automated checks, this could also rely on File I/O to read stimuli and/or produce signal traces that can be compared to reference traces. It would be also good if the HDL simulator output could be captured/checked as well as part of the automated tests.

MarcinOrlowski, 2021-10-04:
> collecting a number of benchmark designs of different complexity, which could be used for manual testing

Just a note that it must be crystal clear what "it works" mean in case of each  of these

BFH-ktt1, 2021-10-04:
Closing issue, please reopen if problem persist

closed 2021-10-04 by BFH-ktt1