logisim-evolution/logisim-evolution #1195: Release V3.7.0

BFH-ktt1, 2021-10-04:
I would, when possible release, at the end of this week V3.7.0, including PR's #1157 , #1184, and #1194.
@R3dst0ne : what is your status on #1157, would you be able to finish it before the end of the week?
@maehne : could you look into #1184 and give me feed-back.

In general: are all in favor, please react with :+1:, the others with :-1:

MarcinOrlowski, 2021-10-04:
Please let me first check if I haven't planted more of #1192

BFH-ktt1, 2021-10-04:
What's the time-frame you think? I'll wait for your okay.

MarcinOrlowski, 2021-10-04:
I believe I should be done by mid-week so we are still on your planned time frame.

maehne, 2021-10-04:
I agree with the overall time frame and try to have a new look at #1184 till Wednesday evening.

maehne, 2021-10-04:
I will also try to have a look on PR #1194 in the same time frame.

maehne, 2021-10-04:
I just got notified that there's a [new release 7.1.0](https://github.com/johnrengelman/shadow/releases/tag/7.1.0) of ShadowJar.

MarcinOrlowski, 2021-10-04:
I'd **not** upgrade for the next release. There's nothing that bites us so "don't fix until it's broken" applies :)

davidhutchens, 2021-10-04:
I agree with the time frame.

I also agree with @MarcinOrlowski about upgrading dependencies. There are actually a number of dependencies that can be updated. I would suggest we hold dependency upgrades until after 3.7.0. After that, we can upgrade them and use the new versions as we build toward the next release. That would give us a chance to fix any issue that might arise from the upgrades. I'll check them out and plan a PR sometime after 3.7.0 is out.