logisim-evolution/logisim-evolution #1196: Project export function and own project management

BFH-ktt1, 2021-10-04:
My PR #1194 implements the project-export function as requested in #13.
How does it work:

1) The user selects a directory where the project should be stored.
2) Logisim, removes the project files of this project if they already exist, and makes in this directory following directories,
  * ```<project name>/libraries``` (<project name> is the name of the main .circ file)
  * ```<project name>/circuit``` (<project name> is the name of the main .circ file)
3) Logisim copies all the external loaded library (```.jar``` and/or ```.circ```) file from their original location to the directory ```<project name>/libraries```.
4) Logisim than saves the project ```<project name>.circ``` file in the directory ```<project name>/circuit```.
5) Note that you now have 2 copies of your project, the original one, and the one saved in ```<project name>/*```.
6) All file names in the new project are now relative to the directory ```<project name>/circuit```

Another advantage, that addresses #63 is that you can also build your own directory hierarchy, as #1194 introduces library relative directory referencing to the ```<project name>.circ```, hence ```../../mydir1/mydir2/mylibrary,jar```.

MarcinOrlowski, 2021-10-04:
> Logisim, removes the project files of this project if they already exist, and makes in this directory following directories,

This should never be the case. If user selected existing folder with the content we should always refuse to use it.

BFH-ktt1, 2021-10-04:
Okay, this we than have to fix in another PR

BFH-ktt1, 2021-10-04:
I commented out the deletion code, have to build an "ask the user" part.

MarcinOrlowski, 2021-10-04:
You may simplify your task by just showing the dialog saying "This folder already exists and is not empty. Please specify non-existing OR empty folder".

MarcinOrlowski, 2021-10-04:
I am afraid this is a blocker for this feature to be merged. Overwrite existing files is as bad as wiping old content. It would be (my guess) best to ask the user what to do (with default action to "Abort" the whole operation) but as there's no much "trafic" here, I think it will be just fine to check if user selected folder exists and simply refuse using it. We can make it more user friendly in subsequent PRs.

BFH-ktt1, 2021-10-05:
I added in my latest push a dialog that asks the user what to do, delete and overwrite or cancel and select a new directory.

BFH-ktt1, 2021-10-05:
Problem has been addressed, and the deletion/overwrite is only performed in case the user wants it.

MarcinOrlowski, 2021-10-05:
I will still not like it. I do not mind overwriting single file - that's pretty much clear for the user what is going to happen when s/he confirm this action. But if you want to wipe content of the arbitrary selected folder (which can be anything really) then it is like stepping on minefield. Users can have similarly named folders and make a mistake, they can have some files there placed by accident so they will confirm annihilation of files they did not intend to loose etc, etc. 

What if we have this feature save all files into **temporary** folder, add some sort of "manifest file" and then pack them as ZIP archive?