logisim-evolution/logisim-evolution #1197: Loaded .circ libraries are not updated in the active logisim session

BFH-ktt1, 2021-10-04:
If you work on a project where you have ```bla.circ``` file loaded as library, this library is not updated if you change in another logisim instance the contents of ```bla.circ```. The only way of updating this library in your original project is to reload it in a new logisim instance (where the file ```bla.circ``` is reloaded). It would be nice if logisim would "monitor" these loaded ```.circ``` libraries for changes and reload them automatically when changes are performed.

MarcinOrlowski, 2021-10-04:
Just a note that this may look less trickier that it really is, as filesystem monitoring is platform specific and may depend on filesystem used.

maehne, 2026-08-21:
I am closing this as resolved by PR #2816. @BFH-ktt1: Please reopen if you think the automatic reloading should still be considered. Personally, I prefer to reload a library explicitly to avoid accidentally breaking the currently open design.

closed 2026-08-21 by maehne