logisim-evolution/logisim-evolution #1198: Saved .circ file contains a lot of library-tool attributes that are "default"

BFH-ktt1, 2021-10-04:
When saving a ```.circ``` file the libraries are written with tool-attributes that are default, example (empty file):

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<project source="3.7.0dev" version="1.0">
  This file is intended to be loaded by Logisim-evolution v3.7.0dev(https://github.com/logisim-evolution/).

  <lib desc="#Wiring" name="0">
    <tool name="Pin">
      <a name="appearance" val="NewPins"/>
    </tool>
    <tool name="Probe">
      <a name="appearance" val="NewPins"/>
    </tool>
  </lib>
  <lib desc="#Gates" name="1">
    <tool name="PLA">
      <a name="table" val=""/>
    </tool>
  </lib>
  <lib desc="#Plexers" name="2">
    <tool name="Multiplexer">
      <a name="enable" val="false"/>
    </tool>
    <tool name="Demultiplexer">
      <a name="enable" val="false"/>
    </tool>
  </lib>
  <lib desc="#Arithmetic" name="3"/>
  <lib desc="#Memory" name="4">
    <tool name="D Flip-Flop">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="T Flip-Flop">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="J-K Flip-Flop">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="S-R Flip-Flop">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="Register">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="Counter">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="Shift Register">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="Random">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="RAM">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool name="ROM">
      <a name="appearance" val="logisim_evolution"/>
      <a name="contents">addr/data: 8 8
0
</a>
    </tool>
  </lib>
  <lib desc="#I/O" name="5"/>
  <lib desc="#TTL" name="6"/>
  <lib desc="#TCL" name="7">
    <tool name="TclGeneric">
      <a name="content">library ieee;
use ieee.std_logic_1164.all;

entity TCL_Generic is
  port(
    --Insert input ports below
    horloge_i  : in  std_logic;                    -- input bit example
    val_i      : in  std_logic_vector(3 downto 0); -- input vector example

	  --Insert output ports below
    max_o      : out std_logic;                    -- output bit example
    cpt_o      : out std_logic_Vector(3 downto 0)  -- output vector example
  );
end TCL_Generic;
</a>
    </tool>
  </lib>
  <lib desc="#Base" name="8">
    <tool name="Text Tool">
      <a name="font" val="SansSerif plain 12"/>
      <a name="halign" val="center"/>
      <a name="text" val=""/>
      <a name="valign" val="base"/>
    </tool>
  </lib>
  <lib desc="#BFH-Praktika" name="9"/>
  <lib desc="#Input/Output-Extra" name="10"/>
  <lib desc="#Soc" name="11">
    <tool name="Rv32im">
      <a name="SocBusSelection" val=""/>
    </tool>
    <tool name="Nios2">
      <a name="SocBusSelection" val=""/>
    </tool>
    <tool name="SocBus">
      <a name="SocBusIdentifier" val="0x0000017C4CE3B25B70a7219d"/>
    </tool>
    <tool name="Socmem">
      <a name="SocBusSelection" val=""/>
    </tool>
    <tool name="SocPio">
      <a name="SocBusSelection" val=""/>
    </tool>
    <tool name="SocVga">
      <a name="SocBusSelection" val=""/>
    </tool>
    <tool name="SocJtagUart">
      <a name="SocBusSelection" val=""/>
    </tool>
  </lib>
  <main name="main"/>
  <options>
    <a name="gateUndefined" val="ignore"/>
    <a name="simlimit" val="1000"/>
    <a name="simrand" val="0"/>
  </options>
  <mappings>
    <tool lib="8" map="Button2" name="Menu Tool"/>
    <tool lib="8" map="Button3" name="Menu Tool"/>
    <tool lib="8" map="Ctrl Button1" name="Menu Tool"/>
  </mappings>
  <toolbar>
    <tool lib="8" name="Poke Tool"/>
    <tool lib="8" name="Edit Tool"/>
    <tool lib="8" name="Wiring Tool"/>
    <tool lib="8" name="Text Tool">
      <a name="font" val="SansSerif plain 12"/>
      <a name="halign" val="center"/>
      <a name="text" val=""/>
      <a name="valign" val="base"/>
    </tool>
    <sep/>
    <tool lib="0" name="Pin">
      <a name="appearance" val="NewPins"/>
    </tool>
    <tool lib="0" name="Pin">
      <a name="appearance" val="NewPins"/>
      <a name="facing" val="west"/>
      <a name="output" val="true"/>
    </tool>
    <sep/>
    <tool lib="1" name="NOT Gate"/>
    <tool lib="1" name="AND Gate"/>
    <tool lib="1" name="OR Gate"/>
    <tool lib="1" name="XOR Gate"/>
    <tool lib="1" name="NAND Gate"/>
    <tool lib="1" name="NOR Gate"/>
    <sep/>
    <tool lib="4" name="D Flip-Flop">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
    <tool lib="4" name="Register">
      <a name="appearance" val="logisim_evolution"/>
    </tool>
  </toolbar>
  <circuit name="main">
    <a name="appearance" val="logisim_evolution"/>
    <a name="circuit" val="main"/>
    <a name="circuitnamedboxfixedsize" val="true"/>
    <a name="clabel" val=""/>
    <a name="clabelfont" val="SansSerif bold 16"/>
    <a name="clabelup" val="east"/>
    <a name="downloadBoard" val=""/>
    <a name="downloadFrequency" val="-1.0"/>
    <a name="simulationFrequency" val="2.0"/>
  </circuit>
</project>
```

where it should be:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<project source="3.7.0dev" version="1.0">
  This file is intended to be loaded by Logisim-evolution v3.7.0dev(https://github.com/logisim-evolution/).

  <lib desc="#Wiring" name="0"/>
  <lib desc="#Gates" name="1"/>
  <lib desc="#Plexers" name="2"/>
  <lib desc="#Arithmetic" name="3"/>
  <lib desc="#Memory" name="4"/>
  <lib desc="#I/O" name="5"/>
  <lib desc="#TTL" name="6"/>
  <lib desc="#TCL" name="7"/>
  <lib desc="#Base" name="8"/>
  <lib desc="#BFH-Praktika" name="9"/>
  <lib desc="#Input/Output-Extra" name="10"/>
  <lib desc="#Soc" name="11"/>
  <main name="main"/>
  <options>
    <a name="gateUndefined" val="ignore"/>
    <a name="simlimit" val="1000"/>
    <a name="simrand" val="0"/>
  </options>
  <mappings>
    <tool lib="8" map="Button2" name="Menu Tool"/>
    <tool lib="8" map="Button3" name="Menu Tool"/>
    <tool lib="8" map="Ctrl Button1" name="Menu Tool"/>
  </mappings>
  <toolbar>
    <tool lib="8" name="Poke Tool"/>
    <tool lib="8" name="Edit Tool"/>
    <tool lib="8" name="Wiring Tool"/>
    <tool lib="8" name="Text Tool"/>
    <sep/>
    <tool lib="0" name="Pin"/>
    <tool lib="0" name="Pin">
      <a name="facing" val="west"/>
      <a name="output" val="true"/>
    </tool>
    <sep/>
    <tool lib="1" name="NOT Gate"/>
    <tool lib="1" name="AND Gate"/>
    <tool lib="1" name="OR Gate"/>
    <tool lib="1" name="XOR Gate"/>
    <tool lib="1" name="NAND Gate"/>
    <tool lib="1" name="NOR Gate"/>
    <sep/>
    <tool lib="4" name="D Flip-Flop"/>
    <tool lib="4" name="Register"/>
  </toolbar>
  <circuit name="main">
    <a name="circuit" val="main"/>
  </circuit>
</project>
```

Need too look into details to clean this up

MarcinOrlowski, 2021-10-04:
Can you please also post the diff?

BFH-ktt1, 2021-10-04:
Here you go:
[diff.txt](https://github.com/logisim-evolution/logisim-evolution/files/7281048/diff.txt)

BFH-ktt1, 2021-10-04:
Note that some of them are "nosave" attributes that are not instantiated like that (for example in the SOC-library)

BFH-ktt1, 2021-10-06:
fixed in #1203

closed 2021-10-06 by BFH-ktt1