logisim-evolution/logisim-evolution #1199: Hide and remove unused libraries on save

BFH-ktt1, 2021-10-05:
My PR #1194 adds a checkbox to the template preferences that says

```Remove unused libraries on save```

If checked logisim will only write those libraries in the ```.circ``` file that are actually used in the projects. This means that when reopening this ```.circ``` file, all unused build-in libraries will be "hidden" and all unused loaded libraries will not be loaded. This helps in cleaning the library browser.

Note that the build-in libraries can be easily made visible again. 

For the moment the default is that this option is not activated. I would like to make this option default. All in favor please react with :+1: , the others with :-1:

MarcinOrlowski, 2021-10-05:
> Note that the build-in libraries can be easily made visible again.

Then `Remove unused libraries on save` sounds misleading and should rather read `Hide unused libraries on load`?

BFH-ktt1, 2021-10-05:
> > Note that the build-in libraries can be easily made visible again.
> 
> Then `Remove unused libraries on save` sounds misleading and should rather read `Hide unused libraries on load`?

Not really, as the libraries to load are "saved" in the logisim ```.circ``` file, hence if "gates" is not in the ```.circ``` file it will not be visible in the library tab :-)

BFH-ktt1, 2021-10-05:
Also the libraries visible at startup depend on the template used (plain, empty, default, or user), it indicates which build-in libraries are visible, which tools are on the toolbar, and which mouse-mappings are done. If you look at #1198, you will see the sections in the beginning with ```<lib> .... </lib>``` the libraries to "show" with their user applied settings for each tool, than ```<toolbar> ... </toolbar>``` with the tools in the toolbar, and than ```<mappings> .... </mappings>``` with the mouse mappings. The same goes for the templates (plain, empty, default, or user)