logisim-evolution/logisim-evolution #1200: Reduce the libraries visible in the library pane in the default template

BFH-ktt1, 2021-10-05:
Currently all build-in libraries (except for the HDL-IP library that is kept for backward-compatibility) are enabled in the default template. For most use-cases only a sub-set are used.

I would like to reduce the libraries in the default template to:
* Wiring
* Gates
* Plexers
* Memory
* Input/Output

All in favor please put a :+1: , all against a :-1:

MarcinOrlowski, 2021-10-05:
Mind explaining why you want this change?

maehne, 2021-10-07:
Personally, I don't have a problem that all libraries are shown by default. This ensures discoverability of all components offered by Logisim-evolution. However, I can understand the wish to reduce the default set shown to user's after first installation, as it makes sense from a didactic point to reduce the choice to a subset, which is recommended for "modern" digital design. Also, we can thus hide libraries, which contain components we consider as outdated/deprecated.

The issue I see is that the dialogue for selecting the visible libraries is currently a bit hidden in the menus. I would be good if it would be part of some sutable tab of the main preferences window.