logisim-evolution/logisim-evolution #1202: Changing circuit Appearance from Custom to Logisim_Evolution, Logisim_HolyCross or Classic Logisim will erase custom appearance from config.

yaybee, 2021-10-05:
For circuits with a Custom appearance it is not possible to switch back to this custom appearance after setting it to one of the standard appearances.
Now custom defaults to Logisim_Evolution, because the switch deleted the complete <appear> tag, however the circuit <a name="appearance" val="custom"/> tag still states it is a Custom appearance.

Solution: when using a standard appearance the <a> tag should be set accordingly (<a name="appearance" val="logisim_evolution"/>), but the custom appearance should be kept for switch-back to it.

BFH-ktt1, 2021-10-05:
I can confirm the finding

MarcinOrlowski, 2021-10-05:
Do we really need to keep the support for other appearances? Can't we just keep LSe and remove all the other?

BFH-ktt1, 2021-10-05:
Yes definitly, as 1) there is no effort to keep them, and 2) it is a service to our users. The custom appearance is a special case and needs to be kept, as you can draw your own. The problem is that the custom appearance disaapears when switching lies in the original logisim code where the "custom" overruled the normal one, hence one's custom there was no way to switch back. I know this problem already several years, but found it too unimportant (compared to other issues) to spend time on it. I will keep this issue open, but as it says on low priority :-)

BFH-ktt1, 2021-10-09:
Closing this issue as addressed by #1206, please reopen if the problem persists.