logisim-evolution/logisim-evolution #1213: Project export function and own project management

BFH-ktt1, 2021-10-09:
My PR #1209 implements the project-export function as requested in #13.
How does it work:

1) The user selects a directory where the project should be stored as ```.zip```-file.
2) If the there already exists a ```.zip```-file with the same name logisim will refuse to overwrite it and the user will have to select another directory.
3) Logisim then creates the ```<project name>.zip```-file (where ```<project name>``` is the name of the main ```.circ```-file) with following contents:
* The main ```.circ```-file is stored at root-level.
* The external loaded library (```.jar``` and/or ```.circ```) are store in the folder ```/libraries```
4) All file names in the ```.zip``` are now relative to the directory ```./```

Another advantage, that addresses #63 is that you can also build your own directory hierarchy, as #1194 introduces library relative directory referencing to the ```<project name>.circ```, hence ```../../mydir1/mydir2/mylibrary.jar```.

MarcinOrlowski, 2021-10-09:
> If the there already exists a .zip-file with the same name logisim will refuse to overwrite

I absolutely cannot understand why this should work differently from regular `save`. If user chooses to export and name of target zip file points to existing file we should ask as what to do and overwrite if chooen so, exactly as we do when s/he saves regular `*.circ` file.

BTW: I missed that previously but localization strings for `Export project` should end up with `...` as this menu item opens another dialog/frame.

BFH-ktt1, 2021-10-09:
> I absolutely cannot understand why this should work differently from regular `save`. If user chooses to export and name of target zip file points to existing file we should ask as what to do and overwrite if chooen so, exactly as we do when s/he saves regular `*.circ` file.

with a regular `*.circ` file it is clear that it is a logisim file, however, `*.zip` could be anything, and in line with your earlier comments I opted for the option to not give the user the possibility of overwrite, as they may overwrite something important.

maehne, 2021-10-12:
I basically agree with the updated description. However, I would recommend to ask the user for each extracted that would potentially overwrite an existing file, whether he actually wants to do that. For the user, it would be helpful if he would get shown basic information (i.e., creation date, last modification date, and file size) for the existing file and the new file. In case of `.circ`files additional information may be displayed like the contained sub-circuits.