logisim-evolution/logisim-evolution #1217: Saved projects' label font is incorrectly shown in properties pane after #1211

MarcinOrlowski, 2021-10-09:
The #1211 comes with known, low-impact issue:

When you have project created with pre #1211 version of LSe, that contains element with a label (i.e TCL) and the label string set (but the label font was not explicitely set upon creation, it is what was the default back then):

[image]

That what you see in old version. When you load this project in post #1211 version you would see this instead:

[image]

which is incorrect as label is rendered in Bold, while properties say it is plain (which is not true). To definitely get rid of it you need to edit the  label which means for old projects this gets out of sync. IIRC this is because we do not save font specs string and fall back to in-app defaults, but we still save some font attributes? And if so, we should rather "regenerate" this string based on object attributes instead of using the default. 

_Originally posted by @MarcinOrlowski in https://github.com/logisim-evolution/logisim-evolution/pull/1211#pullrequestreview-775587082_

maehne, 2021-10-09:
Some users have complained in the past that Logisim-evolution enforces in the GUI and circuits bold fonts. It would be nice if we could make this customisable and ensure that people can override it globally and locally.

BFH-ktt1, 2021-10-10:
The problem is even worst: if you select for example the font `SansSerif Italic`, and save the file, logisim will put `SansSerif plain` in the saved file. Hence the Font issue goes way deeper.

BFH-ktt1, 2021-10-10:
Addendum: the `style` attribute used in saving the font in logisim cannot be manipulated by the font-chooser. This it what caused the initial problem with `bold` (before #1211). Hence I see two possible solutions to the problem:
1) Modify the font-chooser such that instead of `Style` where it now shows the `name` attributes of the given family we have a checkbox of `plain`, `bold`, `italic`. This would make backward compatibility easy, but requires that the currently used `org.drjekyll.fontchooser.FontChooser` is adapted.
2) Drop the `style` attribute from the saved file and use the `name` attribute instead (to be more clear, when looking at a font we have following information: `java.awt.Font[family=SansSerif,name=SansSerif.bolditalic,style=plain,size=16]` where logisim uses in the moment only the `family`, `style`, and `size` field)

I would opt for 1) as it is easier to keep backwards compatible.

closed 2021-10-15 by BFH-ktt1