logisim-evolution/logisim-evolution #1218: Potential security issue

zidingz, 2021-10-09:
Hey there!

I belong to an open source security research community, and a member (@srikanthprathi) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

JamieSlome, 2021-10-10:
@MarcinOrlowski - https://huntr.dev/bounties/a00aa232-4eef-4b09-9138-4896b2fae432/

MarcinOrlowski, 2021-10-10:
Just for a record: https://owasp.org/www-community/vulnerabilities/XML_External_Entity_(XXE)_Processing

MarcinOrlowski, 2021-10-10:
* #1220

kailando, 2026-09-01:
@MarcinOrlowski @BFH-ktt1  Any thoughts on closing this issue with the 4.2 upd?

maehne, 2026-09-04:
@BFH-ktt1: I propose that we close this issue as resolved, as @MarcinOrlowski did not react. He can reopen it if he thinks that there is still something left to be done.

MarcinOrlowski, 2026-09-04:
Sorry, missed that one. +1 for closing.

closed 2026-09-04 by maehne