logisim-evolution/logisim-evolution #1224: I/O pins in 74245

TheByteAttic, 2021-10-10:
I've made my own model of the 74245 bus transceivers. But there is no way to specify that certain pins can be inputs OR outputs, depending on the value of another pin (in this case, DIR). The tool requires that pins either be inputs always, or outputs always. This prevents models of transceivers. Is there a solution to this?

MarcinOrlowski, 2021-10-11:
We shall update the docs and add "Current limitations" list mentioning these few most asked features missing

MarcinOrlowski, 2021-10-11:
Added #1225

* #1225

MarcinOrlowski, 2021-10-11:
We plan to have simulation engine improved to lift this limitation but this is not yet done, co I am closing this ticket as `wont-fix` as this is currently the only option.

closed 2021-10-11 by MarcinOrlowski