logisim-evolution/logisim-evolution #1236: VHDL Circuit doesn't show its I/O port in 3.7.0

diegodutra, 2021-10-15:
The newer version of Logisim-evolution presents a bug in my Ubuntu 20.04 system, as any VHDL circuit I wrote doesn't show its I/O ports. I shall highlight that the same circuit works ok in version 3.6.1. This non-VHDL subcircuit is working ok.

diegodutra, 2021-10-15:
java -jar ./logisim-evolution-3.7.0-all.jar 
java.lang.UnsupportedOperationException: VHDL component simulation is not supported. This could be because there is no Questasim/Modelsim simulation server running.
	at com.cburch.logisim.vhdl.base.VhdlEntity.propagate(VhdlEntity.java:258)
	at com.cburch.logisim.instance.InstanceComponent.propagate(InstanceComponent.java:410)
	at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(CircuitState.java:341)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:272)
	at com.cburch.logisim.circuit.Simulator$SimThread.loop(Simulator.java:338)
	at com.cburch.logisim.circuit.Simulator$SimThread.run(Simulator.java:384)

For 3.6.1, this problem never  occurs

java -jar ./logisim-evolution-3.6.1-all.jar 
java.lang.UnsupportedOperationException: VHDL component simulation is not supported. This could be because there is no Questasim/Modelsim simulation server running.
	at com.cburch.logisim.vhdl.base.VhdlEntity.propagate(VhdlEntity.java:299)
	at com.cburch.logisim.instance.InstanceComponent.propagate(InstanceComponent.java:440)
	at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(CircuitState.java:371)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:289)
	at com.cburch.logisim.circuit.Simulator$SimThread.loop(Simulator.java:357)
	at com.cburch.logisim.circuit.Simulator$SimThread.run(Simulator.java:403)
java.lang.UnsupportedOperationException: VHDL component simulation is not supported. This could be because there is no Questasim/Modelsim simulation server running.
	at com.cburch.logisim.vhdl.base.VhdlEntity.propagate(VhdlEntity.java:299)
	at com.cburch.logisim.instance.InstanceComponent.propagate(InstanceComponent.java:440)
	at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(CircuitState.java:371)
	at com.cburch.logisim.circuit.Propagator.stepInternal(Propagator.java:437)
	at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:300)
	at com.cburch.logisim.circuit.Simulator$SimThread.loop(Simulator.java:357)
	at com.cburch.logisim.circuit.Simulator$SimThread.run(Simulator.java:403)

BFH-ktt1, 2021-10-15:
I can confirm this problem.

BFH-ktt1, 2021-10-15:
Difficult to say. I'm for the 3.7.1, but after #1209 has landed.

diegodutra, 2021-10-15:
@MarcinOrlowski and @BFH-ktt1 thanks.