logisim-evolution/logisim-evolution #1247: PlaRom in subcircuit makes clock stop working

chenzhuoyu, 2021-10-19:
PLA ROM (in section Input / Output Extra) placed in a sub-circuit can stop the clock from working.

Clicking on the clock throws the following exception in console:

```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot invoke "com.cburch.logisim.circuit.CircuitState.getData(com.cburch.logisim.comp.Component)" because "this.circuitState" is null
	at com.cburch.logisim.instance.InstanceStateImpl.getData(InstanceStateImpl.java:56)
	at com.cburch.logisim.std.io.extra.PlaRom.getPlaRomData(PlaRom.java:227)
	at com.cburch.logisim.std.io.extra.PlaRom$Logger.getBitWidth(PlaRom.java:140)
	at com.cburch.logisim.instance.InstanceLoggerAdapter.getBitWidth(InstanceLoggerAdapter.java:48)
	at com.cburch.logisim.gui.log.ComponentSelector$CircuitNode.computeChildren(ComponentSelector.java:236)
	at com.cburch.logisim.gui.log.ComponentSelector$CircuitNode.<init>(ComponentSelector.java:192)
	at com.cburch.logisim.gui.log.ComponentSelector$CircuitNode.computeChildren(ComponentSelector.java:264)
	at com.cburch.logisim.gui.log.ComponentSelector$CircuitNode.<init>(ComponentSelector.java:192)
	at com.cburch.logisim.gui.log.ComponentSelector.setRootCircuit(ComponentSelector.java:463)
	at com.cburch.logisim.gui.log.ComponentSelector.<init>(ComponentSelector.java:389)
	at com.cburch.logisim.gui.log.ComponentSelector.findClocks(ComponentSelector.java:512)
	at com.cburch.logisim.circuit.Simulator.ensureClocks(Simulator.java:583)
	at com.cburch.logisim.circuit.Simulator.setAutoTicking(Simulator.java:540)
	at com.cburch.logisim.gui.menu.MenuSimulate$MyListener.actionPerformed(MenuSimulate.java:379)
	at com.cburch.logisim.gui.menu.MenuItemHelper.actionPerformed(MenuItemHelper.java:45)
	at com.cburch.logisim.gui.menu.MenuItemCheckImpl.actionPerformed(MenuItemCheckImpl.java:29)
	at com.cburch.logisim.gui.menu.LogisimMenuBar.doAction(LogisimMenuBar.java:138)
	at com.cburch.logisim.gui.menu.MenuListener.doAction(MenuListener.java:45)
	at com.cburch.logisim.gui.main.LogisimToolbarItem.doAction(LogisimToolbarItem.java:50)
	at com.cburch.logisim.gui.main.SimulationToolbarModel.itemSelected(SimulationToolbarModel.java:90)
	at com.cburch.draw.toolbar.ToolbarButton.mouseReleased(ToolbarButton.java:85)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
```

Reproducible with circuit:
[pla_in_subcirc.circ.zip](https://github.com/logisim-evolution/logisim-evolution/files/7369937/pla_in_subcirc.circ.zip)

BFH-ktt1, 2021-10-19:
I took a quick peek in the PLA-code from the ITA-branch (IO-extra) and it is badly implemented; the component keeps the `circuitstate` inside the factory, which works when you only have one circuit, but it breaks when you use sub-circuits (each sub-circuit has it's own `circuitstate`). Here there needs quite some rework.

henriquejsza, 2026-08-29:
I reproduced this on current `main`: a root circuit with a subcircuit containing a Clock and the I/O Extra PlaRom throws the same NPE when auto-ticking is enabled (`Simulator.setAutoTicking(true)`).

The immediate crash looks narrower than the state-ownership problem discussed above. During clock discovery, `ComponentSelector.findClocks` asks each loggable component for its bit width through an `InstanceLoggerAdapter` whose metadata state has no `CircuitState`. For PlaRom, `PlaRom.Logger.getBitWidth()` routes through `getPlaRomData()` → `state.getData()`, which dereferences that null state. The output width is already carried by `ATTR_OUTPUTS`, the same attribute that sizes the output port, so reading it directly in `getBitWidth()` avoids the null-state path. With that one-line change, a focused regression test that starts auto-ticking on the nested-PlaRom circuit passes.

To be clear, that would only be an interim fix. It doesn't address how `PlaRomData` mixes saved table contents, simulation values, and editor state, or the question of whether the component should eventually be merged or deprecated.

@hewzhew, I don't want to step on the redesign you offered to plan. If you're not already working on this and the maintainers are fine with the narrow scope, I'd be glad to add the regression test and fix just this clock-startup crash. Would that be helpful?

hewzhew, 2026-08-29:
Thanks for checking and for asking. I'm not currently working on the redesign, so please go ahead with the focused crash fix and regression test if the maintainers agree. Keeping the broader PlaRom state-ownership work separate makes sense to me.

maehne, 2026-09-04:
@henriquejsza: Please go forward with the narrow fix so that it can be included in the upcoming release.

@hewzhew: IMHO, your proposed option 1 "rework the I/O-extra PlaRom by separating saved table contents from per-instance simulation/UI state, while preserving old .circ compatibility as much as possible" is preferable, as both PLA components have different interfaces and attributes making merging them difficult without impacting backwards compatibility.

@BFH-ktt1 and @davidhutchens: What do you think?

davidhutchens, 2026-09-04:
Both @BFH-ktt1 and @hewzhew conclude the component is badly broken. If @hewzhew thinks it can be fixed with reasonable effort, I have no objection. If it proves too difficult, we can fall back to option 2.

hewzhew, 2026-09-05:
Thanks. I will treat Henrique's focused crash fix as the immediate work and keep the broader state-ownership redesign separate. Before committing to option 1, I will first assess the saved-file compatibility boundary and whether separating persistent table contents from per-instance simulation/UI state is reasonably bounded. If it is not, we can still fall back to option 2.

maehne, 2026-09-07:
I am in favour of @davidhutchens proposal on how to move forward in this matter.

hewzhew, 2026-10-01:
Thanks. #3015 is now merged. PlaRom data now refreshes when the shared Contents attribute changes, and cloned simulation states have independent matrix/value arrays. Legacy Contents strings are preserved, with regression coverage for .circ save/reload, two subcircuit instances, resizing, and cloning.

Remaining work:
- The clock-discovery NPE discussed by Henrique: read the output width from the component attribute without requiring a CircuitState.
- Remove the shared editor context from ContentsAttribute and complete the separation of saved table contents from per-instance simulation/UI state.

#1247 remains open for these follow-ups.