logisim-evolution/logisim-evolution #1277: GUI zoom factor reset to auto

mb200002, 2021-11-03:
I observed in Logisim-Evolution 3.7.1 that once the zoom factor for the gui is set manually, there is no possibility to set it back to auto. Maybe a reset button in the preferences window could be inserted.

maehne, 2021-11-09:
I cannot reproduce this behaviour with the HEAD of `develop` (d239fc01). The zoom can be controlled with the controls on the bottom left:

[image]

Clicking on "Auto" after zooming in or out a circuit yields the (at least by me) expected behaviour. If the pane should not be centred on the circuit, you can use the cross hairs button on the bottom right:

[image]

I will close this issue, but feel free to reopen it and explain a bit in more detail the behaviour, which you would expect.

mb200002, 2021-11-09:
Thanks for your explanations. Maybe my description was not too good. I'm adressing the window/gui zoom/scaling factor used for high resolution desplay  (e.g. 4k) accessible over preferences->window. 
[image]

maehne, 2021-11-09:
Sorry @mb200002, I overlooked that you were referring to the zoom factor for the GUI elements! I'm not sure, how useful it still is, as I don't have any 4k monitor, which I can use together with my Windows or Linux laptops for testing. On macOS with a retina screen, I always keep the zoom factor at 1.0 and never had any issues with readability/usability of the GUI. If I remember correctly, there have been also improvements to OpenJDK to deal better with high resolution screeens.

Nevertheless, your request providing a reset functionality for the zoom factor is reasonable. The easiest approach I think would be using the existing reset button for hte window layout on the same reference pane.

BFH-ktt1, 2022-08-26:
Can this issue be closed?

closed 2022-08-31 by BFH-ktt1

mb200002, 2022-09-08:
@BFH-ktt1 @maehne 
Tested Nightly #663 from 20220831 using .deb pkg on Ubuntu 20.04.5 LTS, looks good  to me.