logisim-evolution/logisim-evolution #1280: Create and edit the appearance of a sub-circuit in v3.7.1 doesn't work as expected

chenzhuoyu, 2021-11-04:
Expected behavior:

1. Open a `.circ` file created with a version before v3.7.1
2. Create a new sub-circuit
3. Edit it's appearance
4. Add the sub-circuit into main circuit, it will show the edited appearance

What happened instead:

In step 4, it always shows the default appearance.

This does not happen in v3.7.0.

If I then open the circuit in v3.7.0 and edit it's appearance, it will show that somehow the circuit has 2 origins, which is not suppose to happen (I guess).

[image]

If I remove the long rectangle and it's origin, and re-open it in v3.7.1, everything returns normal.

I guess somehow the new version is not handling the "origin" properly?

chenzhuoyu, 2021-11-04:
Also, If I move any pins when editing it's appearance, it won't update in the main circuit (the graphics will update, but not the pins), until I drag the "sub-circuit" or reload the entire project.

I don't know if this is the same problem as above?

BFH-ktt1, 2021-11-07:
Closing as it is fixed in #1281 . Please reopen if the problem persists.

chenzhuoyu, 2021-11-07:
@BFH-ktt1 I have tested the latest `develop` branch, the pin-movement bug disappear, but I still cannot edit subcircuit's appearance, and the duplicated-origin still exists when I open in v3.7.0

maehne, 2021-11-08:
Reopen due to remaining issue mentioned by @chenzhuoyu.

BFH-ktt1, 2021-11-08:
@maehne , @chenzhuoyu : I cannot reproduce this behavior in develop. Please be more specific on "but I still cannot edit subcircuit's appearance" and say with which version of "logisim-evolution". That it cannot be done in V3.7.0 and V3.7.1 is clear as they still contain the bugs.

maehne, 2021-11-08:
I just tested editing the custom appearance with HEAD (f96b702f) from branch `develop`. It works for me (see [my_mux.zip](https://github.com/logisim-evolution/logisim-evolution/files/7495404/my_mux.zip)):

[image]

Though, I noted that the default custom appearance when starting editing: 

[image]

does not correspond to the default Logisim-evolution appearance for the block:

[image]

@BFH-ktt1: Is this intented behaviour? If yes, I think we can close this issue.

BFH-ktt1, 2021-11-08:
@maehne : Yes this is intended behavior, the standard "custom appearance" before editing is just the port's, the anchor, and a square. I opted for this such that when editing the custom appearance you do not have to "delete" too much elements.

closed 2021-11-08 by BFH-ktt1

chenzhuoyu, 2021-11-08:
@BFH-ktt1 @maehne Please don't close this so quickly ...

I did a screen recording to show what I mean, hope this describes better ...

https://user-images.githubusercontent.com/8230799/140718836-f15eb61c-426b-4db2-8609-275923a5bf3e.mp4

chenzhuoyu, 2021-11-08:
As we can see, no matter how I edit the appearance of `foo`, it always show as the "default" look in main circuit. This (I guess) is not the intended behavor.

BFH-ktt1, 2021-11-08:
@chenzhuoyu : I looked into your video, and the behavior is as it should, as you have set the circuit "foo" the appearance to "logisim-evolution", and not to "custom". As the editing of the appearance will not automatically switch to "custom appearance" as it used to be (due to an issue that people wanted to have the ability to switch between the different appearances) you have to switch your circuit manually to "custom appearance", and than it will show as custom in your super-circuits.
Hope this helps

chenzhuoyu, 2021-11-09:
@BFH-ktt1  Oh... I never noticed that. Switching to "custom appearance" solves my problem, but this is very confusing, the new circuit was created with it's appearance set to "Logisim-Evolution", I never changed anyting.

Probably it's better to default to "Custom" when creating new circuits to maintain a consistent behavior with old versions?

Also, out of curiosity, although v3.7.0 is buggy and it shows 2 origins, why after I remove the "builtin" one, it then  immediately behaves what I was expecting in v3.7.1?

dusklight00, 2022-09-02:
Thank you so much! I totally did not saw the appearance section.

Sammot, 2022-11-24:
Please just make the custom appearance the default, it took me ages to work out why It wasn't showing my custom appearance and it seems a rather needless option.

txz32102, 2023-06-03:
> Please just make the custom appearance the default, it took me ages to work out why It wasn't showing my custom appearance and it seems a rather needless option.

so how to make the custom appearance the default, it made me suffer a lot

antoinemayol, 2023-07-26:
Thanks you! Same for me, it tooks a while to figure out this ;)

Enprogames, 2023-11-09:
Where is this "custom appearance" selection? I too have customized my appearance, but cannot figure out how to switch from the default.

Enprogames, 2023-11-09:
Never mind haha. It's in the bottom under the "Properties" tab in the bottom-left pane. There is a dropdown menu "Appearance", and you can change it from "Logism-Evolution" to "Custom".