logisim-evolution/logisim-evolution #1295: NPN transistor is broken

loggeek, 2021-11-13:
Hi,

It seems like the NPN transistor is broken, since when both the source & the gate are set, the output is floating even throught it should be set as well.

Here's a side by side comparison with the original Logisim (Evolution at left, Logisim at right):
[image]

Thanks!

davidhutchens, 2021-11-14:
Thank you for the report. This issue is similar to others raised before (#431, #504). Logisim-evolution has the correct digital behavior. Note that the model is a simplified digital model of transistors for specific digital purposes. There are many uses of transistors that the simulation does not handle. 

In digital usage N-Type transistors can drive the output to 0, but cannot drive the output to 1. A P-Type transistor can drive the output to 1, but cannot drive it to 0. Hence, in your case, the output should be floating (U).

To see proper use in practice, look at typical CMOS NOT, NAND, and NOR circuits. You'll see the N-Type transistors connecting ground (0) to the output and P-Type transistors connecting power (1) to the output, with the logic such that exactly one side drives the output at any given time. It doesn't work to put the P-Type on the ground side or the N-Type on the power side because they cannot effectively drive that direction.

By the way, Logisim-evolution provides transistors to allow instructors to describe typical logic gate implementations. We discourage their use in more complex circuits. Use logic gates instead.

Does @maehne or @BFH-ktt1 have anything to add or correct in my response?

loggeek, 2021-11-14:
OK, understand now. Thanks for the explanation!

closed 2021-11-14 by loggeek