logisim-evolution/logisim-evolution #1297: Cannot display component names in Design panel in CJK languages

myl7, 2021-11-17:
As the app ships Chinese and Japanese translations, when choosing the two languages, while other places show the translation correctly, the component names in Design panel just disappear and leave it blank.

I guess it is maybe a font problem though I am not sure.
I have tried to set locale envs like `LC_ALL` and `LANG`, but they donot help.

Here are some screenshots:

English works fine:

[image]

Japanese and Chinese break:

[image]
[image]

Environment:

- Version: 3.7.2
- OS: Arch Linux
- JDK: openjdk version "17.0.1" 2021-10-19

R3dst0ne, 2021-11-19:
The current head of develop works fine for me.
[image]

- OS: Kubuntu 20.04
- Java:
  - openjdk 17 2021-09-14
  - OpenJDK Runtime Environment (build 17+35-Ubuntu-120.04)
  - OpenJDK 64-Bit Server VM (build 17+35-Ubuntu-120.04, mixed mode, sharing)
  
  This could be a case of Arch being Arch. Unfortunately, none of the core developers uses Arch on a regular basis, so we can't give you specific advice.
  Can you check if the locales for Chinese and Japanese are installed?
  Can you try to build the current head of develop yourself and see if it throws any errors when the locale get changed?

myl7, 2021-11-19:
> Can you check if the locales for Chinese and Japanese are installed?

Yes, `cat /etc/locale.gen | sed '/^#/d'` outputs:

```
en_US.UTF-8 UTF-8
ja_JP.UTF-8 UTF-8
zh_CN.GBK GBK
zh_CN.UTF-8 UTF-8
```

And for `cat /etc/locale.conf` if it can help:

```
LANG=en_US.UTF-8
LC_TIME=zh_CN.UTF-8
```

> Can you try to build the current head of develop yourself and see if it throws any errors when the locale get changed?

I have tried latest commit 8c6076df6 of branch develop , which gives the same problem.
No error / warning is throwed out in the compilation `./gradlew shadowJar`.

R3dst0ne, 2021-11-19:
Try `./gradlew run` and then change the locale.
If if this doesn't throw an error I unfortunately can't help you.

myl7, 2021-11-19:
`./gradlew run` still gives the problem, throwing no error/warning

maehne, 2021-11-24:
@Narita-1095305, @legion-null, and @Unknown-User2: Could someone of you maybe help to find the cause of this issue? I suspect the mostly likely cause is the use of or fallback to fonts, which don't include glyphs for Chinese/Japanese characters.

Unknown-User2, 2021-11-24:
i,m not on arch anymore im on gentoo now

legion-null, 2021-11-24:
我认为有可能，你应该尝试一下安装常用的cjk字体

I think it is possible, you should try to install commonly used cjk fonts
另：推荐使用文泉驿开源字体
PS:Recommend to use Wenquanyi open source font

myl7, 2021-11-24:
Normal CJK fonts have been installed already, including wqy-zenhei wqy-microhei noto-fonts-cjk. It is kinda wired that the program doesnot fallback to use them.

maehne, 2025-07-30:
I am sorry, but I cannot reproduce the issue on my side with the development HEAD on macOS. Chinese and Japanese locales work fine without any tofu placeholders displayed in the Library pane. I suspect that on your system, a font gets used, which lacks the Chinese characters. Having CJK fonts installed won't help if they don't get used by Logisim-evolution. From our side this is hard to debug. It would be great if someone could investigate it, who can reproduce the issue reliably.