logisim-evolution/logisim-evolution #1298: VHDL Simulator Restart

pbuccella, 2021-11-18:
The "Restart VHDL simulator" command does not seem to be effective in the case errors are encountered in the vhdl code after a "VHDL Simulator Enabled". In this case to restart the simulation, Logisim must be closed and reopened. is this a bug?

maehne, 2021-11-18:
@pbuccella: Thanks for the report! This certainly sounds like a bug. Could you please provide a reproducer? Ideally, one where in the VHDL code, you mark with comment the region with and without error.

pbuccella, 2021-11-19:
Hi, thank you for you answer. Here yo find a simple VHDL code example with the commented region with and without error. The file I have sent with an error.

[Test_bug_1298.zip](https://github.com/logisim-evolution/logisim-evolution/files/7568731/Test_bug_1298.zip)

esseivan, 2021-12-07:
I've looked a bit into it while debugging and found that it didn't restarted because it is blocked on the state "Starting" in the function "com.cburch.logisim.vhdl.sim.VhdlSimulatorTop.restart()". More specifically, it stops at "com/cburch/logisim/vhdl/sim/VhdlSimulatorTclBinder.java:110" and for some reason doesn't seems to reach the line 145 (so 149 which reset the state) of the same file.

maehne, 2021-12-09:
Thanks @esseivan for your analysis! I hope we find the time to fix it in the next release. A PR is welcome!

esseivan, 2021-12-11:
I looked a bit yesterday but found no solution.
This morning I had the idea that maybe it's just waiting for the readLine instruction to be completed.
I found https://stackoverflow.com/questions/7855822/bufferedreader-readline-method-hangs-and-block-program#7855911 that may confirms it.
still, I don't know how to fix it other than doing a timeout like a waitAny

maehne, 2022-08-23:
@pbuccella and @esseivan: Can this issue be closed since PR #1326 got merged?

maehne, 2022-10-26:
Thanks for the positive feedback @esseivan! I close therefore the issue as resolved.

PR #1326 (2022-01-14): Fix for fail to restart when vhdl code invalid
closes #1298
closed 2022-10-26 by maehne