logisim-evolution/logisim-evolution #1303: VHDL "Simulation halted by internal error"

finallyfound, 2021-11-25:
Hi team,

I am encountering an issue, when trying to simulate some VHDL code:

logisim (version 3.7.2) states the message in title with following specifics:

Thanks for looking into this...

# 10.1d

# do ../run.tcl 50294 
Run simulation environnment build...
# ** Warning: (vlib-34) Library already exists at "work".
# 
# Modifying modelsim.ini
# Model Technology ModelSim ALTERA vcom 10.1d Compiler 2012.11 Nov  2 2012
# -- Loading package STANDARD
# -- Loading package TEXTIO
# -- Loading package std_logic_1164
# -- Compiling entity LogisimVhdlSimComp_0
# -- Compiling architecture TypeArchitecture of LogisimVhdlSimComp_0
# Model Technology ModelSim ALTERA vcom 10.1d Compiler 2012.11 Nov  2 2012
# -- Loading package STANDARD
# -- Loading package TEXTIO
# -- Loading package std_logic_1164
# -- Loading package NUMERIC_STD
# -- Compiling entity top_sim
# -- Compiling architecture comp of top_sim
# ** Error: (vsim-3296) Unknown option '-errorfile'.

# Use the -help option for complete vsim usage.
# Error in macro ./../run.tcl line 31
# ** Error: (vsim-3296) Unknown option '-errorfile'.

# Use the -help option for complete vsim usage.
# 
#     while executing
# "vsim top_sim"

maehne, 2021-11-25:
Thanks for the report. Could you please provide a complete self-contained test case in form of a ZIP archive allowing us to reproduce your issue on our side? Additionally, please specify, which version of Logisim-evolution, Intel Quartus Prime, and operating system you are using. Maybe your ModelSim version is just a bit too old.

PS: Please, use [Markdown formatting](https://docs.github.com/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax) correctly in future issue reports.

finallyfound, 2021-11-25:
thanks for your comments
here are the requested details:

OS: windows 10
JRE version 17.0.1
logisim-evolution 3.7.2 (jar version)
quartus II 13.1 (this was the last version supporting Cyclone III FPGA used with DE0 board)

[TC_issue_1303.zip](https://github.com/logisim-evolution/logisim-evolution/files/7604616/TC_issue_1303.zip)

maehne, 2021-11-25:
Quartus II 13.1 is quite old as is the accompanying ModelSim version. Even though you depend on the old Quartus II version, you can still install a newer ModelSim version for simulation. Could you, e.g., give the ModelSim-Intel FPGA Starter Edition shipped with Quartus Prime lite 18.1 a try? It is available as a separate download on [Intel's web site](https://fpgasoftware.intel.com/18.1/?edition=lite&platform=windows).

finallyfound, 2021-11-26:
thanks for this hint. It is working now (used 18.1 as recommended)

closed 2021-11-26 by finallyfound