logisim-evolution/logisim-evolution #1306: ROM and RAM raw import errors

nailed-barnacle, 2021-11-26:
I have a C-generated .raw file consisting of 16k of 64-bit integers. When loaded into the ROM or RAM part I get a single bit error, but only in certain circumstances.

Logisim-evolution V3.7.2, Linux Mint 20.0.2 (also occurs with 3.7.1 installed via flatpack)
barnacle@barnacle-Latitude-7480:~/Projects/Logisim-Evolution/fake8080/Debug$ java --version
openjdk 11.0.11 2021-04-20
OpenJDK Runtime Environment (build 11.0.11+9-Ubuntu-0ubuntu2.20.04)
OpenJDK 64-Bit Server VM (build 11.0.11+9-Ubuntu-0ubuntu2.20.04, mixed mode, sharing)

The test files are located at my Github repository here: https://github.com/nailed-barnacle/Fake8080 (I don't seem to be able to attach them here).

1) when uploading seq.raw, the word at address 0x0a87 (line 60 of the text file) should be 0x00003f76 but appears as 0x00003f56. This also occurs at 0x0aa7, which replicates the first 32 bytes of line 60. However, it is correctly loaded at addresses 0x0007 and 0x0027...

2) seqx.raw is a file made by copying and pasting the offending line 60 from seq.raw 256 times, to replicate any invisible characters (just in case!) - when that is loaded into RAM or ROM, it loads correctly in every location.

The workaround is to manually change the word after each load, but that does rather defeat the object of the exercise... it's doing my head in. I can't think of any obvious reason why apparently only those two words are being thumped, and I don't speak Java so I don't have a clue where to start looking.

Neil

davidhutchens, 2021-11-26:
Thank you for the report. You can upload files here if you zip them. Version 3.7.x requires at least Java 16. Could you upgrade your Java version and see if the problem still occurs? If it does, please zip your files and drop that in your response.

nailed-barnacle, 2021-11-26:
OK, I updated to openJDK 16.0.1 and the issue sadly remains
[broken import.zip](https://github.com/logisim-evolution/logisim-evolution/files/7609359/broken.import.zip)
.
The attached zip contains two .raw files: seqx.raw which works and seq.raw which fails, and for each of those a screen grab of the LE hex editor and of an eclipse editor with the offending word highlighted.

Thanks for the fast response!
Neil

davidhutchens, 2021-11-26:
I think you are looking at the wrong lines. The image showing the problem is on the line starting with address 0a80. That corresponds to line 44 of the raw file. 10*16*16+8*16 = 2688. The raw data has 64 words per line. 2688 / 64 is exactly 42. The first line of the file is the header. The second line corresponds to address 0. So the data for address 0a80 is on line 44 of the file. That line has the data shown in the image, so it seems to be loading correctly.

Please let me know if I am not understanding the issue.

nailed-barnacle, 2021-11-26:
Aaaah! You are quite correct and I am an utter idiot... I misread my own handwriting and fell down a rabbit hole of my own creation.

I apologise unreservedly for wasting your time, and thank you again for your very fast response.

Please log this one as 'no such issue'.

Neil

davidhutchens, 2021-11-26:
I am happy that the problem is resolved. This issue can be closed.

closed 2021-11-26 by nailed-barnacle