logisim-evolution/logisim-evolution #1308: Delayed text edit when "Use Questa Advanced Simulator to validate HDL entites" is enabled

esseivan, 2021-11-27:
When the feature "Use Questa Advanced Simulator to validate HDL entites" is enabled, typing in the VHDL entity editor takes some times instead of being instant.
As well, the CPU usage while typing goes up by about 15% (compared to no change when it is disabled).
This occurs wether or not this entity is used in a circuit.

My guess is that's a bug because there no real-time validation as far as I know.

I'm using logisim evolution v3.7.1 with modelsim 20.1

maehne, 2021-11-28:
Thanks for the report! Could you please also state on which OS version you are working?

esseivan, 2021-11-28:
Sure, here it is 👍 
Edition	Windows 10 Home
Version	21H2
Installed on	‎05/‎08/‎2020
OS build	19044.1348
Experience	Windows Feature Experience Pack 120.2212.3920.0

esseivan, 2021-12-07:
From what I could find, it is due to the fact that at "com/cburch/logisim/vhdl/base/VhdlContent.java:289", the validation is called each time the document is edited, because the function "setContent" is also called.
I don't know if that is intended, but personally I would rather have an option to disable it and only validate when clicking the validate button.

BFH-ktt1, 2021-12-15:
@esseivan : Can this Issue be closed now that #1327 has been merged?

esseivan, 2021-12-15:
Yes, all good for me

PR #1327 (2021-12-15): No validation of VHDL code while editing it
closes #1308 

New function that does not validate the code which is called when necessary
closed 2021-12-15 by esseivan