logisim-evolution/logisim-evolution #1317: Mystery ghost ticks incrementing a counter.

alsilisk, 2021-12-05:
hello, 

I appear to be having issues with ghost ticks and a counter circuit. 

Attached is my project. The lowest signal on the scope is the clock signal to the 24 bit counter to the left of the scope. It only triggers once and is rising edge triggered. Yet it increments twice. 

Is this a software bug or is it an issue on my part? 

https://drive.google.com/drive/folders/1HyMefgTrwUZhGiSyhY4UTqPQYrXO5w-u?usp=sharing

maehne, 2021-12-05:
The counters in your design are not directly connected to the clock source. The issue you are observing is due to the gating of the clock signal, which causes hazards that yield additional rising clock edges to which your counters react. Instead of gating the clock, it is more robust to directly connect the clock source to the clock inputs and use the logic signal, which was used for the gating as enable signal for the counter.

closed 2021-12-05 by maehne

alsilisk, 2021-12-05:
appreciated, thank you