logisim-evolution/logisim-evolution #1318: LEDs of RGB LED matrix on LEGUAN FPGA board stay on in certain corner cases

maehne, 2021-12-07:
We have a simple traffic light controller FSM example, which we use when introducing the topic in our course (see [support_slides_week7.zip](https://github.com/logisim-evolution/logisim-evolution/files/7667619/support_slides_week7.zip)). If the sub-circuit `mealy_hyra` is downloaded on the new LEGUAN board, it does not work properly when the RGB LED matrix is used for displaying the traffic light. Ther red and yellow lights always stay on. This example worked well in the past on our old board GECKO4-EDUCATION.

BFH-ktt1, 2021-12-11:
The problem here is that the on- and off-color of those LED's are equal, and that the intensity of both the on- and off-color are >128. This will lead in the current implementation that the LED always stays on (hardwired to 1).
There are several solutions to this problem:

1) The easiest: Generate a warning to the user.
2) The more intelligent: Ask the user what to do (for example take black for the off-state)
3) Ideally and most extensive: Implement a PWM that can "emulate" the intensity on the RGB LED-Array.

maehne, 2021-12-12:
A warning would be a good start. Options 2 or 3 could be implemented at a later point.