logisim-evolution/logisim-evolution #1321: XNOR logic gate wrong output (3 inputs)

gbzinsheikk, 2021-12-09:
Recently I've observed an error when trying to use the XNOR logic gate in logisim-evolution. With three entries, it gives an output error in the case that all three entries are 1(high logic level), it gives an output of one, when it should be zero, accordingly with the truth table of this gate.

esseivan, 2021-12-10:
There's an option "Multiple-Input Behavior" when selecting the xor / xnor gate. To have a real xor you want to select the 2nd option in this parameter.
I believe this 2nd option should be the default one.

BFH-ktt1, 2021-12-11:
@gbzinsheikk : Basically a XOR/XNOR gate with more than 2 Inputs do not exists, as there are than two definitions for the behavior, either 1) react only when 1 input is active (the default behavior) , or 2) react when an odd/even number of inputs is active (parity generation).
I always suggest to do a cascading of XOR/XNOR gates when more than 2 inputs are required, or use alternatively the Odd/Even Parity blocks if the other functionality is required.
Closing this issue as the behavior currently in logisim-evolution is correct.

closed 2021-12-11 by BFH-ktt1