logisim-evolution/logisim-evolution #1322: Counter Max Value saved correctly, but not load

magical-obama, 2021-12-10:
When I save this project, the maximum values for the counter get saved, but when I close **Logisim**, these values reset to **0x0**, even though they are saved in the project!

I don't know what to do

[image]
[image]

davidhutchens, 2021-12-10:
Your .circ file has the width (i.e. the number of data bits) set to 7. But you are giving a maximum value of 0x80 which requires 8 bits. Thus the 8th bit is dropped and it loads 0. The max value should be no more than 0x3f if you have 7 bits.

I'm not sure how you set it that way. What version of Logisim-evolution are you using? When I try to set that in version 3.7.1 (setting bits to 7 and max to 0x80, in either order) it immediately changes the max value to 0 and then stores it in the file that way.

magical-obama, 2021-12-10:
[image]

https://user-images.githubusercontent.com/72757458/145607169-6a90dfb3-7e62-49f0-8b32-bdd9c5cdb041.mp4

davidhutchens, 2021-12-10:
Interesting. I can verify the behavior you show in 3.7.2. However, neither 3.7.1, nor the head of the current develop branch, nor the head of the master branch permit that setting. So I don't think there is a current bug to fix. But we need to be aware not to re-introduce it when merging branches.

@BFH-ktt1, do you have any explanation for this behavior?

BFH-ktt1, 2021-12-11:
@magical-obama , @davidhutchens : I find it an interesting issue, as I cannot reproduce it on linux in 3.7.2. I guess it has to do with the java version that is shiped with the .msi-file (will check this). @magical-obama : What happens if you fill in for example 0x81, or 0x91. Both should be out of range for the 7-bits and should be corrected to 0x01, resp. 0x11.

davidhutchens, 2021-12-11:
I think I can replicate this issue in Java 16 and 17 and in 3.7.2 and head of develop! The trick is to be persistent. The first try drops the 8 but the second try lets 0x80 be set. It alternates truncating and not as you continue to type values. If you use 0x91 it truncates to 0x11 and then stores 0x91 on the next try.

Now that I can reproduce it, I can look for the cause.

davidhutchens, 2021-12-11:
I think I understand where the code is wrong. It is a bit convoluted (which is why this problem exists), so I should try to clean it up also. I'll prepare a PR when I have something I am happy with.

maehne, 2021-12-11:
Thanks @davidhutchens for looking into this issue!

davidhutchens, 2022-01-03:
This issue is resolved by #1346. It is no longer possible to force a Max value that is out of range.

This issue can now be closed.

PR #1346 (2022-01-03): Fix Max update in Counter and problems with undo when extra attributes are changed
This PR closes #1322. It also fixes the attribute undo problem described in #1337.
closed 2022-01-03 by BFH-ktt1