logisim-evolution/logisim-evolution #1324: Simulation no longer works.

BrianRushka, 2021-12-11:
I hope this is an operator error on my part. I have only used logisim-evolution for about 10 days.

After simulating several different circuits, suddenly simulation no longer works: poke, clock, etc. 
Cannot set value of inputs. Please see attached window image.

Attempted:
1) Uninstalling and reinstalling logisim-evolution.
2) re-boot PC

Running Rel 3.7.2
OS: Feature update to Windows 10, version 21H1
      Successfully installed on ‎12/‎8/‎2021

[image]
.

davidhutchens, 2021-12-11:
You seem to be using the classic Logisim appearance for enable and gatedClock. But the classic appearance for an output is a circle. It looks like you have gatedClock as an input facing west. That would cause a conflict and it wouldn't run as expected. If that is the problem, change it to an output and reset the simulation. If you use the Logisim-evolution arrow appearance, the direction may be more clear.

Let me know if that works. If I am missing something, please zip your circuit and attach the file here so we can examine it.

BrianRushka, 2021-12-11:
[simplest.zip](https://github.com/logisim-evolution/logisim-evolution/files/7696257/simplest.zip)
I expect to see the value of the inputs/outputs and a color to indicate the state of each wire. None of these is
apparent.
Further troubleshooting: I attempted the same circuit in old logisim 2.7.1 and see same issue!

davidhutchens, 2021-12-11:
The simplest.circ that you attached above seems to work for me.

Did you start with a complex circuit that wasn't working and stripped it down to this simple circuit? Did you start with a fresh launch of Logisim-evolution and open just the simplest.circ to see if that works? I am guessing that your complex circuit has caused a problem in the simulation that is not clearing when you strip out things. So try a fresh launch of LE and open just simplest.circ. Does that work for you or not?

BrianRushka, 2021-12-11:
Yes. I had problems with a more complex circuit.  So, I relaunched LE. Issue remained. Next, I uninstalled LE and re-installed. Built a simple circuit which exhibited the problem. Then I inspected my OS updates to see if the problem appeared after an update of Windows.  I have rebooted my PC.  I have not tried reverted my OS yet.

As I mentioned above, I installed original logisim 2.7.1. 

Besides Java runtime, any other components which might be persisting data?

BTW, thanks so much for looking into this!

davidhutchens, 2021-12-11:
I'm not seeing a problem. Can you attach the complex circuit that fails.

BFH-ktt1, 2021-12-11:
@BrianRushka : You have, maybe by a keystroke, switched logisim to the mode "printer view", in which you do not see the simulation colors. Please go to Window->Preferences->Layout and untick the second option "Printer View"; this should do the trick.
Hope this helps.

BrianRushka, 2021-12-11:
That is it, precisely!  Everything is fine now.Do you need me to close the issue, or follow up in any way?
Much thanks to both of you for you kind attention.-Brian

BFH-ktt1, 2021-12-11:
@BrianRushka : You're welcome. Closing issue.

closed 2021-12-11 by BFH-ktt1

tpolecat, 2022-09-16:
Wow this just happened to me too. Glad this issue was here. I think the issue may be that the whitespace to the right of the checkbox controls is part of the controls themselves, so if you click in the empty area by mistake you'll toggle the controls. Just a guess.

[image]

maehne, 2022-09-16:
@tpolecat: The printer view option has been removed by PR #1486 to resolve the related issue #1437. So, this should not happen anymore in the upcoming release. If you think the active area causes more issues, please open a new issue so that we don't forget about it.