logisim-evolution/logisim-evolution #1325: Error if "work" exists but is empty

esseivan, 2021-12-11:
I had an error for the last day had no idea where it came from. Reinstalling completely logisim and modelsim did not helped.

I looked into it and found that in the temp path used, there was an empty "work" folder. So modelsim thinks the library exists, reads what's into it but found nothing, so returns an error.

I think the program could look into the folder and check if the required library files are present, or delete the whole folder.

This issue appears only when "Use Questa Advanced Simulator to validate HDL entities" is enabled.

[image]

maehne, 2021-12-13:
Thanks @esseivan for reporting this issue! Looking into the `work` directory for ModelSim files may become quickly fragile, because the design library format is not documented and may change from one ModelSim/QuestaSim version to another. A more robust way needs to be implemented. Maybe, ModelSim's `vdir` command offers the required functionality. We may also reconsider creating the `work` design library as a physical folder `work/`. Since `work` is just an alias to the current design library, it is actually not recommended, though supported by ModelSim, to call the physical design library `work` when creating it using the `vlib` command. The current `work` library is selected using the `vmap` command. The design libraries and their mappings are stored in the `modelsim.ini` file, which is usually found in the directory, from which the ModelSim commands are launched. A PR addressing the issue is welcome!

primechaser, 2022-10-24:
Hi both, 
I am running into the same issue and disabling Questa simulator is not an option for me. Is there a hack to somehow get around this?

In my case (windows 10) "work" directory is getting updated with external simulator disabled. When enabling it doesn't work anymore. I see modelsim.ini is getting copied into ..\comp directory with updated "work" entry, but for some reason this might not get read....

esseivan, 2022-10-24:
Looks like I forgot to mention it in my report, but my workaround was to delete the said empty work folder. Appdata\Local\Temp\work in my case

primechaser, 2022-10-25:
works now...
thanks @esseivan