logisim-evolution/logisim-evolution #1330: Add Some Common CPU Built (Or All CPU And MCU)

cheznewa, 2021-12-14:
Hello , Is Possible To Add CPUs Built (Not As Circuit) Like i386 ARM ... ?
Docs ::::: https://github.com/larsbrinkhoff/awesome-cpus

BFH-ktt1, 2021-12-15:
The SOC library provides all the means, so PR's are welcome!

maehne, 2021-12-15:
The SoC library already provides instruction set simulators for RISC V and Nios II. If you would like to see other instruction set architectures to be supported, please be more specific and, ideally, open a pull request. Implementing and maintaining support for new processor architectures requires effort and therefore should be driven by a concrete demand.

For the moment, I consider your question answered. Feel free to reopen it to make your request more precise.

closed 2021-12-15 by maehne