logisim-evolution/logisim-evolution #1331: Upgrade dependencies

maehne, 2021-12-15:
Some of our dependencies are outdated:

```
$ ./gradlew dependencyUpdates

> Task :dependencyUpdates

------------------------------------------------------------
: Project Dependency Updates (report to plain text file)
------------------------------------------------------------

The following dependencies are using the latest milestone version:
 - at.swimmesberger:swingx-core:1.6.8
 - com.github.ben-manes.versions:com.github.ben-manes.versions.gradle.plugin:0.39.0
 - commons-cli:commons-cli:1.5.0
 - javax.help:javahelp:2.0.05
 - net.sf.nimrod:nimrod-laf:1.2
 - org.apache.commons:commons-text:1.9
 - org.drjekyll:colorpicker:1.3
 - org.hamcrest:hamcrest:2.2
 - org.mockito:mockito-inline:4.1.0
 - org.mockito:mockito-junit-jupiter:4.1.0
 - org.scijava:swing-checkbox-tree:1.0.2

The following dependencies have later milestone versions:
 - com.fifesoft:rsyntaxtextarea [3.1.3 -> 3.1.4]
     http://www.fifesoft.com/rsyntaxtextarea/
 - com.formdev:flatlaf [1.6.4 -> 1.6.5]
     https://github.com/JFormDesigner/FlatLaf
 - com.github.johnrengelman.shadow:com.github.johnrengelman.shadow.gradle.plugin [7.1.0 -> 7.1.1]
 - com.puppycrawl.tools:checkstyle [8.45 -> 9.2]
     https://checkstyle.org/
 - de.jflex:jflex [1.4.1 -> 1.8.2]
     https://jflex-de.github.io/jflex/
 - org.junit:junit-bom [5.8.1 -> 5.8.2]
     https://junit.org/junit5/
 - org.junit.jupiter:junit-jupiter [5.8.1 -> 5.8.2]
     https://junit.org/junit5/
 - org.slf4j:slf4j-api [1.7.32 -> 2.0.0-alpha5]
     http://www.slf4j.org
 - org.slf4j:slf4j-simple [1.7.32 -> 2.0.0-alpha5]
     http://www.slf4j.org

Gradle release-candidate updates:
 - Gradle: [7.3 -> 7.3.2]

Generated report file build/dependencyUpdates/report.txt
```

Especially, the [shadow plugin](https://github.com/johnrengelman/shadow) got recently updated to [version 7.1.1](https://github.com/johnrengelman/shadow/releases/tag/7.1.1) due to  [CVE-2021-44228](https://github.com/advisories/GHSA-jfh8-c2jp-5v3q) and [CVE-2021-45046](https://github.com/advisories/GHSA-7rjr-3q55-vv33).

Maybe, we should consider using [Dependabot](https://docs.github.com/en/code-security/supply-chain-security/keeping-your-dependencies-updated-automatically/about-dependabot-version-updates) for automatically updating our dependencies.

R3dst0ne, 2022-07-03:
@maehne since #1355 got merged, can this be closed?

closed 2022-08-23 by BFH-ktt1

maehne, 2022-09-23:
@BFH-ktt1: I would have left this issue open, as it also suggests automatising the dependency updates using Dependabot.