logisim-evolution/logisim-evolution #1333: Program freezes when using Compose key in Edit Circuit Layout

flarn2006, 2021-12-20:
I just ran into this annoying bug in Logisim Evolution v3.7.2, running on Arch Linux through java-17-openjdk. I tried using the Compose-`~`-`~` sequence to type a `≈` character in a label, and the program stopped responding completely, causing me to lose what I was working on.

Through further testing, I determined that it freezes whenever any valid Compose sequence is typed inside of a label on the circuit layout, whether you're editing a label on a component or a bare label created with the text tool.. This applies only to Edit Circuit Layout; the Compose key works as intended for labels in Edit Circuit Appearance. It also works in regular text fields (including properties) as well as the Keyboard component.

**To reproduce:**

1. Run Logisim Evolution on Linux after enabling a Compose key.
2. Select the text tool and click anywhere in the window.
3. Enter a Compose key sequence (e.g. `Compose`-`a`-`").

**Expected result:** The character associated with the provided Compose sequence (`ä`, if you used the provided example) should be typed.

**Actual result:** The program freezes and needs to be forcibly terminated.

MarcinOrlowski, 2021-12-21:
>  set the Compose key to the right Alt key.

Mind be more specific what are the exact steps here?

flarn2006, 2021-12-22:
@MarcinOrlowski It varies depending on your OS setup, but it's usually in the keyboard settings. I'm using KDE Plasma, and for me it's in System Settings → Keyboard → Advanced tab → Position of Compose key.

[image]