logisim-evolution/logisim-evolution #1334: TTL model woes: 74139 has inverted output logic, 7485 HDL generator outputs the wrong HDL code.

raps500, 2021-12-20:
These findings apply to version 3.7.2:

The 74139 model has inverted logic, when an output is selected it should be at 0 and the others at 1:
`
    for (var i = 0; i < 4; i++) {
      final var val = enabled ? (outputPortStates[A + B][i] == 1 ? Value.TRUE : Value.FALSE) : Value.TRUE;
      state.setPort(mapPort(outPorts[i]), val, DELAY);
    }

`

The 7485 HDL generator outputs wrong HDL kind: if verilog it outputs VHDL and vice-versa:

`    if (Hdl.isVerilog()) {
      contents.addVhdlKeywords().add("""
            oppA   <= A3&A2&A1&A0;
            oppB   <= B3&B2&B1&B0;
            gt     <= '1' {{when}} unsign`

maehne, 2021-12-31:
@MarcinOrlowski: Could you maybe have a look to this issue?

MarcinOrlowski, 2022-01-01:
@raps500 can you please verify the issue is still present in latest stable or even develop build?

raps500, 2022-01-02:
As of 02.01.2022, both the main and develop branches still have both bugs, is there any other branch/version I can test ? The last activity on the nighty was 11 days ago.

MarcinOrlowski, 2022-01-02:
`develop` is development head, so that's sufficient. Thanks

quick0831, 2022-03-20:
I just try to do some experiment on 74139

[74LS139 datasheet](https://www.futurlec.com/74LS/74LS139.shtml)

Screenshot of the datasheet:
[image]
[image]

Test Circuit:
[image]

Timing Diagram:
[image]

quick0831, 2022-03-20:
Seems like the inputs are ok
only need to invert the output

quick0831, 2022-03-20:
Sorry I missed that.
I also checked the [datasheet](https://www.ti.com/product/SN74LS139A) mentioned in code.

It still said the same thing

[image]

(I tested in `develop` branch)

quick0831, 2022-03-20:
I tried to put some fix (tested with same *.circ file)
[image]

better

quick0831, 2022-03-24:
https://github.com/logisim-evolution/logisim-evolution/blob/master/src/main/java/com/cburch/logisim/std/ttl/Ttl7485HdlGenerator.java#L59-L59

For this line mentioned by OP, it should be changed to `Hdl.isVhdl()`

https://github.com/logisim-evolution/logisim-evolution/blob/master/src/main/java/com/cburch/logisim/fpga/hdlgenerator/Hdl.java#L39-L45

maehne, 2022-03-28:
@quick0831: I agree with your proposed fix. Could you please make a PR?

BFH-ktt1, 2022-08-31:
Closing, as PR was merged

PR #1433 (2022-04-13): 7485HdlGenerator fix
fix #1334
closed 2022-08-31 by BFH-ktt1